sandeepmistry/arduino-BLEPeripheral #103: Differences in RFduino and BLE Nano ANCS

danlsk, 2016-11-11:
I made the same code for RFduino and BLE Nano ANCS. I remove the call for cleardata() from the example on both platform to make them the same since only BLE Nano has problem in OTA. There are some different behavior. 

1. RFduino shows as Not Connected under My Devices after power cycle. It will automatically show as Connected when the power come back. There is no need to repair again manually.
2. BLE Nano drop off from My Devices to Other Devices after power cycle. It require manual pairing after power cycle. 

Is this the difference of S110 and S130? The only difference for both platform is the soft device.

sandeepmistry, 2016-11-14:
> Is this the difference of S110 and S130? 

Not sure, maybe you can turn on debug logging? https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/src/nRF51822.cpp#L42

>  It require manual pairing after power cycle.

Just to confirm, you are not reflashing the sketch in this step?

danlsk, 2016-11-20:
I am not reflashing the sketch after the power cycle or reset. I can't find anything with debug logging.

danlsk, 2016-11-23:
This issue relates to omitting clearData(). I managed to get clearData() back to work on BLE Nano. I believe RFduino able to get connected even after omitting clearData() could be due to the fact that previously I have enable clearData() and the pairing bond is in the phone. While BLE Nano has not been able to perform clearData() previously, it doesn't save the pairing bond correctly. At this point, I think this issue can be closed.

sandeepmistry, 2016-11-26:
Closing at mentioned above.

closed 2016-11-26 by sandeepmistry