sandeepmistry/arduino-BLEPeripheral #111: Awesome Library!

MarqueIV, 2016-12-06:
Man, this library is great! I found it thanks to your book, 'Make: Bluetooth'.  Didn't realize you were the author of both until I started digging around.  This is so much cleaner/easier than trying to use RedBearLab's or Nordic's SDKs.  Really well thought out.

Question though... I'm very new to this (but ramping up quickly thanks to your book and library!) and I see code that specifies a read/write characteristic, then a separate notification-supporting characteristic.  Is it possible (and if so, is it advisable) to also mark the read/write characteristic as supporting notifications?   For instance...

Instead of this...
`BLECharCharacteristic switchCharacteristic = BLECharCharacteristic("FF11", BLERead | BLEWrite);
BLECharCharacteristic stateCharacteristic  = BLECharCharacteristic("FF12", BLENotify);`

Can you do this?
`BLECharCharacteristic switchCharacteristic = BLECharCharacteristic("FF11", BLERead | BLEWrite | BLENotify);`

Again, I don't know if that's possible, but it seems like it would make sense considering you can already read the characteristic, so why not get notified when it changes?

sandeepmistry, 2016-12-08:
@MarqueIV thanks for the positive feedback!

> Can you do this?
> `BLECharCharacteristic switchCharacteristic = BLECharCharacteristic("FF11", BLERead | BLEWrite | BLENotify);`

Yes, that works quite well :)

The led switch example uses a similar concept but read + notify: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/led_switch/led_switch.ino

I'm going to close this issue for now, since I think I've answered your question. However, let me know if it's not clear.

closed 2016-12-08 by sandeepmistry

MarqueIV, 2016-12-08:
Glad to hear! That's what I thought.

BTW, been using your ANCS example, but I'm having trouble with reconnections.  While when I first compile, download and run the program, then pair with my iPhone everything works, but simply restarting the board (unplugging then reconnecting it again) doesn't reconnect to the phone. I have to manually 'forget' the device, restart the device, then re-pair it to get it to work again.  Is there some trick to getting it to simply reconnect after being powered off?  Again, I've just started my work from your example here while using your library from the IDE's library manager.

sandeepmistry, 2016-12-18:
> BTW, been using your ANCS example, but I'm having trouble with reconnections. While when I first compile, download and run the program, then pair with my iPhone everything works, but simply [...]

Please take a look at: https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/102#issuecomment-262451748

> Also, these aren't really 'issues' of course. They're just quesitons. Is there a different/better way to communicate with you about your work? i.e. forum, email, etc.?

Github issues or Gitter is fine: https://gitter.im/sandeepmistry/arduino-BLEPeripheral