sandeepmistry/arduino-BLEPeripheral #114: How to enter low-power sleep mode?

markleavitt, 2016-12-08:
Sandeep, thank you for this amazing library that allows those with limited C-prog skills to use the nRF Bluetooth chips!  But I need help now figuring out how to put the chip into its low-power sleep mode between Bluetooth events.

I have an oscilloscope/currentsense setup to measure real-time power consumption and am using a Sparkfun nRF52832 breakout board with the power LED disabled. I've tested both the iBeacon and led_button examples from your repo. I see the expected ~20 mA power consumption spikes from the advertising packets (for the iBeacon) and/or the connection events (for the led_button).  But in between those events, there's a steady 8 mA current draw that would drain a coin cell in just one day. This indicates the chip is not "going to sleep" between Bluetooth events.  This key feature of Bluetooth Low Energy is really important for my wearable devices.

I've used the RFDuino and Simblee chips and their library has an ULPDelay() function that puts the chip to sleep for a specified time, or until a Bluetooth event occurs.  I've confirmed that it works with my measurement setup.  But their modules are proprietary so I would like to use your library instead for my experiments. Is there some function call in your library that I've missed that puts the chip in low power sleep? What does the .poll() method do?  Thanks for your great work and help!

sandeepmistry, 2016-12-18:
@markleavitt 

> Is there some function call in your library that I've missed that puts the chip in low power sleep?

This trying does not attempt to handle any low power scenarios.

A low power sketch for a RedBearLab BLE Nano board can be found here: https://gist.github.com/sandeepmistry/6a6dfe9021b19bfddd60

> What does the .poll() method do?

It checks the Nordic stack for BLE/radio events.

closed 2016-12-18 by sandeepmistry

benlhy, 2017-07-21:
Using this sketch with the nRF52832 board resulted in Arduino throwing the error:

"sd_nvic_EnableIRQ" was not declared in this scope.

cheeta1, 2017-10-30:
i get the same error @benlhy  have you found any solution for nrf52832

FloStar3000, 2020-04-01:
Hi, I use the Sparkfun NRF52832 breakout and I get the same error as the guys above, I tried importing nrf_nvic.h, but that still does not help.