sandeepmistry/arduino-BLEPeripheral #122: Pinouts for Adafruit Bluefruit  LE SPI Friend

aventurella, 2017-02-01:
https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/Eddystone/EddystoneUID/EddystoneUID.ino

The example here shows:
```
// define pins (varies per shield/board)
//
//   Adafruit Bluefruit LE   10, 2, 9
//   Blend                    9, 8, UNUSED
//   Blend Micro              6, 7, 4
//   RBL BLE Shield           9, 8, UNUSED
```

Looking at the Bluefruit LE SPI Friend the names don't all line up and I don't have enough knowledge to know how they map:

https://learn.adafruit.com/introducing-the-adafruit-bluefruit-spi-breakout/introduction

Available SPI Pins:
SCK
MISO
MOSI
CS
IRQ

I am trying to fulfill:
```
#define EDDYSTONE_BEACON_REQ  
#define EDDYSTONE_BEACON_RDY  
#define EDDYSTONE_BEACON_RST  
```

On the SPI Friend Board:
```
RST is RST (Marked on the board) 
RDY is IRQ?
REQ is ???
```

aventurella, 2017-02-01:
This might be answered here:

https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/34
https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/69

AKA out of luck

sandeepmistry, 2017-02-02:
@aventurella that's correct unfortunately.

If you have a J-Link, CMSIS-DAP, or STLink flasher you can try to flash it from the Arduino IDE using: https://github.com/sandeepmistry/arduino-nRF5

I'll add a warning to the read me to mentioned those boards from Adafruit are NOT supported.

closed 2017-02-02 by sandeepmistry