sandeepmistry/arduino-BLEPeripheral #125: Weird syntax when instantiating classes

too1, 2017-02-10:
Hi

Is there a particular reason you instantiate classes like this:
BLEPeripheral blePeripheral = BLEPeripheral();
 
Instead of the more common:
BLEPeripheral blePeripheral; 

?

I would expect the two to be functionally equivalent, but the second might involve an additional object being created if I understand this post correctly:
http://stackoverflow.com/questions/2722879/calling-constructors-in-c-without-new

Since this code is easily visible in all the sketches it would be nice to follow the standard C++ norm.

Best regards
Torbjørn

sandeepmistry, 2017-02-13:
Hey @too1,

I was mostly of styling reasons just to get services and characteristics lined up. I'm not opposed to changing the formatting, feel free to open a pull request. When this library was ported to [CurieBLE](https://github.com/01org/corelibs-arduino101/tree/1.0.7/libraries/CurieBLE) I think the style was changed to the one you suggested.

I quickly tried changing the LED example, the compiled output size was the exact same, so I don't expect anything weird to be going on.

@don what do you think?

sandeepmistry, 2017-04-17:
Closing this for now due to lack of feedback.

closed 2017-04-17 by sandeepmistry