sandeepmistry/arduino-BLEPeripheral #126: ESP32 support

vshymanskyy, 2017-02-17:
This is just an idea/question - would it be very hard to support ESP32 BLE?
It this BLE stack relevant or it would be almost impossible to implement?

Thanks @sandeepmistry !

vshymanskyy, 2017-02-17:
Here is some early stage example for ESP32 BLE advertisement:
https://gist.github.com/me-no-dev/00bc1a4765350e46992bfd9d5d6c1473

sandeepmistry, 2017-02-21:
Hey @vshymanskyy,

ESP32 should be pretty straightforward to add, the library already has a `BLEDevice` abstraction layer  that separates nRF8001 and nRF51/52 support, so it's just a matter of creating a new lower level for ESP32.

There's some higher level API's in the ESP IDF that can be used instead of the link you provided above: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth

Unfortunately, this is not something I have time to do, however I'd gladly review a pull request for this.

jvkassi, 2019-01-08:
@sandeepmistry can you have a look now ?