sandeepmistry/arduino-BLEPeripheral #127: sd_app_evt_wait only waits until central is connected

acaballero, 2017-02-28:
Hello,

I'm trying to put a BLENano board to slepp every loop iteration. These are te steps:

1. setup: 

```
     [...]
     sd_power_mode_set(NRF_POWER_MODE_LOWPWR);    
     NRF_POWER->TASKS_LOWPWR = 0;
     NRF_POWER->DCDCEN = 0x00000000;
     [...]
```

2: loop:

 ```
    [...]

    sd_app_evt_wait();

    BLECentral central = ble.central(); 

    // if a central is connected to peripheral:
    if (central) {        

         sd_app_evt_wait();

        // while the central is still connected to peripheral:
        while (central.connected()) {
               sd_app_evt_wait()
               [...]
     }

    [...]

```

The first time sd_app_evt_wait is called, it waits until a central is connected. But then, the execution enters the inner loop and  sd_app_evt_wait returns instantly.

I've read things about pending events or interrupts, bu I've not managed to get to work. I've also tried calling sd_nvic_EnableIRQ(SWI2_IRQn); with a proper empty SWI2_IRQHandler, but then sd_app_evt_wait never returns and no central can connect to the device.

Please, ¿is there something with the library I have to setup in order for this to work?

Thanks!

sandeepmistry, 2017-03-02:
Hey @acaballero,

It's better to use the poll style API with event handlers as many library calls check for SD events internally.

Examples can be found here: https://gist.github.com/sandeepmistry/6a6dfe9021b19bfddd60

closed 2017-03-02 by sandeepmistry

jeanmatthieud, 2019-06-13:
I have the same issue.
I read that if the IRQ is not enabled, the flag is not cleared and so the sd_app_evt_wait returns instantly.

I tried to enable SWI2_IRQn, and I added an empty 'SWI2_IRQHandler()' (as provided in the gist example).
The device is still visible, but when I try to connect to it, it crashes (no more visible, connection fails...).
I don't know (yet) how to dump the crash informations.

It looks like the SWI2_IRQHandler() is not accessible / executed. Any idea ?