sandeepmistry/arduino-BLEPeripheral #132: BLEHIDPeripheral setBondStore

llemtt, 2017-03-15:
Hi,

how can I correctly set the BondStore of a BLEHIDPeripheral?

If I call bleHIDPeripheral.setBondStore(...) the BLEPeripheral class method gets called which doesn't update the _bleBondStore private field. Calling bleHIDPeripheral.setup() then overrides my previous call...

Thanks

sandeepmistry, 2017-03-26:
@llemtt why do you need to override the BondStore?

llemtt, 2017-03-27:
My current project runs on a BLE Nano and requires a different flash page for the BondStore because offset=0 overwrites the bootloader!

I managed this modifying the library source code, definitely not the right way to do it.

Besides that, if I understood correctly, setBondStore(...) purpose is to let the developer supply even a completely different store location like for instance an external eeprom.

sandeepmistry, 2017-04-09:
Oh got it, why don't we just change the default offset on the BLE Nano to be non-zero?

What do you think? Do you have time to submit a pull request?

sandeepmistry, 2017-04-24:
@llemtt ping ...