sandeepmistry/arduino-BLEPeripheral #135: Dynamic data not saved unless disconnected from central

prossel, 2017-03-21:
**Symptoms**

Using with RedBearLab BLE Shield v 2.1 on Arduino Uno, the peripheral cannot reconnect after first successful connection. 
Unless is it disconnected first from central without being powered down. That is the only way I have found to trigger the save of dynamic data in EEPROM.

**To reproduce**

- Open HIDKeyboard example
- Adjust BLE_REQ and BLE_RDY pins to specific board configuration
- Upload to Arduino
- Connect to central (tested with MacBook, OS X 10.11.6)
- Test from Serial Monitor by sending a few keys, they come back through the HID. (To break the loop, click somewhere where a few keystrokes are not harmful) 
- Disconnect the Arduino (power down)
- Reconnect Arduino
- See that it does not reconnect

**More information**

To make it work, after connecting to the central, use the Bluetooth menu to disconnect the HIDKeyboard or turn off Bluetooth. Do not power off the Arduino. When the connection is lost on Arduino, we can see in the serial monitor that the dynamic data is read (from the shield) and stored to EEPROM (to see these messages, on has to activate debug traces in nRF8001.cpp, search for #define NRF_8001_DEBUG).

I wonder why theses dynamic data is only saved when the peripheral is disconnected from host and not when the bond is successfully established. Also I don't see the ACI_EVT_BOND_STATUS message come when the peripheral is connected (and bond) for the first time.

sandeepmistry, 2017-03-27:
@prossel this is a limitation of the nRF8001 as far as I understand.

Let me know if you have any ideas on how we can avoid this.

prossel, 2017-03-28:
I would suggest to try requesting the dynamic data after the connection has been successfully established, without waiting for the central to disconnect. Maybe nRF8001 support could help find a way.

sandeepmistry, 2017-04-10:
> I would suggest to try requesting the dynamic data after the connection has been successfully established, without waiting for the central to disconnect.

@prossel I don't think this is possible, have you tried it out yourself. If so, please submit a PR.

prossel, 2017-04-10:
@sandeepmistry I tried but with no success.

sandeepmistry, 2017-04-16:
Ok, we'll just leave this issue open to track it as a known limitation.