sandeepmistry/arduino-BLEPeripheral #136: nrF52 and serial example, Connection failed: no serial service

bossesand, 2017-04-05:
I use the serial example, have only changed the BLE_pins to pins not used on the Taida Centurty nrF52

I can see the BLE device in the Android Serial Bluetooth Terminal app but cannot connect
I get the error Connection failed: no serial service
I loaded the S132 to the board first then I flash the app using ST-Link V2

Any pointers on how to proceed?

sandeepmistry, 2017-04-10:
@bossesand do you have a link to the ` Android Serial Bluetooth Terminal` app? Is it intended to be used with the Nordic UART profile?

bossesand, 2017-04-10:
Hi
Here is the BLE nordic UART Apps links that works with adafruit but not
with your example.
I have a Taide Century mini devkit

https://play.google.com/store/apps/details?id=com.nordicsemi.nrfUARTv2
https://play.google.com/store/apps/details?id=com.cmd.blechat

BUT I have not succeded in getting the example to work...

But as I have downloaded the Arduino support for this board I get the
Examples for the feater uart to work, I can send bleuart strings
https://www.adafruit.com/product/3406

BUT a LARGE BUT, I cannot get the Wire example to read a adafruit bno055
module over i2c/TWI and send any sensible data over bleuart.

I have a salea 8 logic analyzer, and I cannot get see any sensible data
reads on the i2c bus, I can see just a lot of NAC accesses.
I saw one time data as it should look but I cannot repeat that.

So I hope you can improve your example so that I can get bleuart
connectivity, I think/hope your Wire implementation is better....

Bo-Erik

bossesand, 2017-04-11:
Hi again
I feel so stupid...  I cannot understand what is going on.
I am modifying a sketch to test the wire implementation with reading a
bno055 module.
I have am using a sketch that I had tried to get to work with BLEPeriferal,
I have removed all references / calls to BLEPeriferal.
Why is the BLEperiferal library still used ? at this moment it might not be
a problem as when I remembered to specify the softdevice S132 the sketch
compiled OK...
But it would be interesting to know.

Can anyone explain how libraries are handle in the arduino windows version?
any link to a description
/user/bosse/Appdata/...
Documents/arduino/library
and libraries in the sketch folder

"C:\Users\bosse\AppData\Local\Arduino15\packages\SparkFun\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-objcopy"
-O binary
 "C:\Users\bosse\AppData\Local\Temp\arduino_build_662850/V2T2-serial-sandeep-Mod.ino.elf"
"C:\Users\bosse\AppData\Local\Temp\arduino_build_662850/V2T2-serial-sandeep-Mod.ino.bin"
"C:\Users\bosse\AppData\Local\Arduino15\packages\SparkFun\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-objcopy"
-O ihex
 "C:\Users\bosse\AppData\Local\Temp\arduino_build_662850/V2T2-serial-sandeep-Mod.ino.elf"
"C:\Users\bosse\AppData\Local\Temp\arduino_build_662850/V2T2-serial-sandeep-Mod.ino.hex"
Using library BNO055 in folder:
C:\Users\bosse\Documents\Arduino\libraries\BNO055 (legacy)
Using library Wire at version 1.0 in folder:
C:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\libraries\Wire
Using library CmdMessenger at version 4.0.0 in folder:
C:\Users\bosse\Documents\Arduino\libraries\CmdMessenger
Using library SPI at version 1.0 in folder:
C:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\libraries\SPI
Using library BLEPeripheral at version 0.4.0 in folder:
C:\Users\bosse\Documents\Arduino\libraries\BLEPeripheral

2017-04-10 17:14 GMT+02:00 Bo-Erik Sandholm <bossesand@gmail.com>:

bossesand, 2017-04-11:
I get so tired of my lacking understanding of this :-(
I recieved a Adafruit nrf52 feather today
https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/examples
I tried to lightly modify the uart example...
'
I want to read a bno055 over i2c and print out quaternion data like this, a
2 as first char for CmdMessenger message type strings
2,0.644,-0.762,-0.012,-0.067;
2,0.644,-0.762,-0.012,-0.067;
without this: !!!
[BLE]: BLE_GAP_EVT_TIMEOUT

I had for a while ble uart connectivity working even if the data was coming
over as one character per line!

When trying to do adhoc stuff to fix that, I lost ble uart connectivity...
so now i do not get uart connectivity to the Android app UART-LE

I just want to run a generic nrf52832 module, read quaternion data from a
bno055 over i2c
and send it over a BLE NUS uart connection to a lot of different master
devices Windows OSX IOS and Android

So a Simple working example to start again with :-)

Bo-Erik

2017-04-11 11:07 GMT+02:00 Bo-Erik Sandholm <bossesand@gmail.com>: