sandeepmistry/arduino-BLEPeripheral #137: multiple addHID instance problem

claudioarduino, 2017-04-07:
Hi all, I try to use the HID_test example from arduino-BLEPeripheral library for interfacing with my IPhone. The problem is that The program work properly if I instance only one addHid:
`bleHID.addHID(bleMultimedia);` 
but when I try to add the keyboardHID:
`bleHID.addHID(bleMultimedia);` 
`bleHID.addHID(bleKeyboard);`
The system does not work, it seem in loop in setup() function or the system in in continous reset.
If somebody had this kind of problem, please help me.
Thanks

sandeepmistry, 2017-04-10:
@claudioarduino what board are you using? It could be a RAM limitation ....

claudioarduino, 2017-04-10:
@sandeepmistry I'm using the Arduino mega2560. I have the same problem also with the ancs example. When I try to add different remote attributes it doesn't work anymore. It works only with one `addRemoteAttribute`. How could it be possible?

bojh, 2017-04-10:
@claudioarduino I do not know how many attributes do want to create. I had a similar problem, adding service characteristics. I found that there is a native softdevice call (e.g. `sd_ble_gatts_characteristic_add`)  in nRF51822.cpp, where the return value is not yet evaluated and have found that my application has reached a (preallocated) memory limit. At Nordic devzone, I have found that there is a limit with 1.5kB
So, the first test would be, if that is the case in your context.
@sandeepmistry .... I can do a PR to create a DEBUG message in that cases.
Here is the snippet to check on errors for adding characteristics:
```
uint32_t ret=sd_ble_gatts_characteristic_add(BLE_GATT_HANDLE_INVALID, &characteristicMetaData, &characteristicValueAttribute, &this->_localCharacteristicInfo[localCharacteristicIndex].handles);
if (ret!=NRF_SUCCESS) {
  Serial.print("sd_ble_gatts_characteristic_add ERROR: "); 
  Serial.print(characteristic->uuid()); Serial.print(": ");
  switch (ret) {
  case NRF_ERROR_INVALID_ADDR:  Serial.println("INVALID_ADDR: Invalid pointer supplied.");	  break;
  case NRF_ERROR_INVALID_PARAM: Serial.println("INVALID_PARAM: Invalid parameter(s) supplied, service handle, Vendor Specific UUIDs, lengths, and permissions need to adhere to the constraints.");  break;
  case NRF_ERROR_INVALID_STATE: Serial.println("INVALID_STATE: Invalid state to perform operation, a service context is required.");  break;
  case NRF_ERROR_FORBIDDEN:     Serial.println("FORBIDDEN: Forbidden value supplied, certain UUIDs are reserved for the stack.");	  break;
  case NRF_ERROR_NO_MEM:        Serial.println("NO_MEM: Not enough memory to complete operation.");         break;
  case NRF_ERROR_DATA_SIZE:     Serial.println("DATA_SIZE: Invalid data size(s) supplied, attribute lengths are restricted by @ref BLE_GATTS_ATTR_LENS_MAX.");      break;
  }
}
```
... same has to be done for services an descriptors to be on the save side at development time.

claudioarduino, 2017-04-10:
@bojh I'm using the nrf8001. I  wanted to use both keyboard and multimedia, but it doesn't work. The native softdevice call  `sd_ble_gatts_characteristic_add`  I can use it also with the nrf8001?

bojh, 2017-04-10:
@claudioarduino nrf8001 has a completely  different implementation, because it acts as a external chip and is realized to tell the configuration in sent messages. I'm not deep inside, but you should check if there is a chance to get, if the creation is accepted or an error has occurred. "Serial" debugging in the library source does help in such cases, to explore what happens really.

claudioarduino, 2017-04-10:
@bojh Thanks for your answers. I enabled the nrf8001 debug but when I try to add more than one attribute the system doesn't work and I'm not able to see the serial debugging. I think there must be something to change.

bojh, 2017-04-10:
@claudioarduino I think you have to add your own Serial.println() commands to see what happens.

claudioarduino, 2017-04-10:
In the HID_test example, in void setup(),  if i put both 
 `bleHID.addHID(bleKeyboard);` 
 `bleHID.addHID(bleMultimedia);` 
 and then
`bleHID.begin();`
`Serial.println(F("BLE HID"));`
in the serial monitor it doesn't print "BLE HID" but stops working after `bleHID.begin();`.
Instead if I put only
`bleHID.addHID(bleMultimedia);`
it works and prints on the serial monitor "BLE HID".