sandeepmistry/arduino-BLEPeripheral #139: How to update the library BLEPeriferal higher than 0.3.1 in Arduino 1.8.1 ?

bossesand, 2017-04-12:
I cannot get the Arduino 1.8.1 to update BLEPeriferal higher than 0,3,1 
I have downloaded the arduino-BLEPeriferal-master.zip  but when I try to add as a zip library I cannot really see anything happening.

It is probably an easy step but I have difficulty remembering how I updated to 4.0 before, and now I backed down as I wanted to remove the library and reinstall it with the latest examples.

cheeta1, 2017-04-15:
hi @bossesand i have faced this issue you have to remove this line

//custom boards may override default pin definitions with BLESerial(PIN_REQ, PIN_RDY, PIN_RST)
BLESerial BLESerial();

and add the following lines

#define BLE_REQ 5
#define BLE_RDY 4
#define BLE_RST 6

BLESerial BLESerial(BLE_REQ, BLE_RDY, BLE_RST);

now it will work

sandeepmistry, 2017-04-16:
IDE 1.8.2 lists v0.4.0 of the library is available (and installed) for me, I suggest you post something on forum.arduino.cc

[image]

closed 2017-04-16 by sandeepmistry