sandeepmistry/arduino-BLEPeripheral #140: Howto compile BLEPeriferal serail example ? I do not succed

bossesand, 2017-04-12:
I am starting to look like a pest... Sorry
I have tried with arduino 1.8.1 and 1.8.2
examples > BLEPeriferal > serial
Board Taide Century and S132

Any idea of what is wrong, cans someone compile the example and send src and hex to me ?

Compiling sketch...
"C:\Users\bosse\Documents\ArduinoData\packages\SparkFun\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os  -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=16000000 -DARDUINO=10802 -DARDUINO_STCT_NRF52_minidev -DARDUINO_ARCH_NRF5  -DNRF52 -DS132 -DNRF51_S132 -DNRF5 "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/gcc/" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/device/" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/softdevice/s132/headers/" -DUSE_LFXO "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\variants\Taida_Century_nRF52_minidev" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\libraries\SPI" "-IC:\Users\bosse\Documents\Arduino\libraries\BLEPeripheral\src" "C:\Users\bosse\AppData\Local\Temp\arduino_build_700153\sketch\BLESerial.cpp" -o "C:\Users\bosse\AppData\Local\Temp\arduino_build_700153\sketch\BLESerial.cpp.o"
"C:\Users\bosse\Documents\ArduinoData\packages\SparkFun\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os  -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=16000000 -DARDUINO=10802 -DARDUINO_STCT_NRF52_minidev -DARDUINO_ARCH_NRF5  -DNRF52 -DS132 -DNRF51_S132 -DNRF5 "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/gcc/" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/device/" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/softdevice/s132/headers/" -DUSE_LFXO "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\variants\Taida_Century_nRF52_minidev" "-IC:\Users\bosse\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.3.0\libraries\SPI" "-IC:\Users\bosse\Documents\Arduino\libraries\BLEPeripheral\src" "C:\Users\bosse\AppData\Local\Temp\arduino_build_700153\sketch\serial.ino.cpp" -o "C:\Users\bosse\AppData\Local\Temp\arduino_build_700153\sketch\serial.ino.cpp.o"
C:\Users\bosse\Documents\Arduino\libraries\BLEPeripheral\examples\serial\serial.ino: In function 'void setup()':

serial:29: error: request for member 'setLocalName' in 'BLESerial', which is of non-class type 'BLESerial()'

   BLESerial.setLocalName("UART");

             ^

serial:32: error: request for member 'begin' in 'BLESerial', which is of non-class type 'BLESerial()'

   BLESerial.begin();

             ^

C:\Users\bosse\Documents\Arduino\libraries\BLEPeripheral\examples\serial\serial.ino: In function 'void loop()':

serial:36: error: request for member 'poll' in 'BLESerial', which is of non-class type 'BLESerial()'

   BLESerial.poll();

cheeta1, 2017-04-15:
hi @bossesand  i have faced this issue you have to remove this line

//custom boards may override default pin definitions with BLESerial(PIN_REQ, PIN_RDY, PIN_RST)
BLESerial BLESerial();

and add the following lines

#define BLE_REQ   5
#define BLE_RDY   4
#define BLE_RST   6

BLESerial BLESerial(BLE_REQ, BLE_RDY, BLE_RST);

now it will work

sandeepmistry, 2017-04-16:
Fixed in https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/f21a2ba925650a2ce90352caa783bc0e95d5e3fe.

closed 2017-04-16 by sandeepmistry