sandeepmistry/arduino-BLEPeripheral #141: Example needed of private GATT service

bossesand, 2017-04-13:
Sorry to disturb again.  I need a new example :-) or pointer
I found that I need to change my goal, I need to send IMU quaternion data 4 signed 16 bit values to a host at around 50 Hz.

I want to pair the nrf52832 with my host, The functionality I need to understand is how to create a private gatt service that can send the 4 16 bit variables to the host.

I can read  the quaternions from the  bno055 over i2c / twi.

I tried to use bleuart, but the overhead to to convert to Strings and format to fixed reolution got too complex.

It seems simpler to not go that roundabout way if possible.

The example would probably fit perfectly to Arduino Primo Core if it is ever released, 
There was leaked info that this aurdino will be nrf52832 + bno055

bossesand, 2017-04-13:
This compiles but is the BLE Service and Characteristics setup correctly?
How about specify IMU instead of Temp ?

Is there any mistake on how to setup a BLEService with 4 signed 16 bit parameters?
- The logic for sending to host at 50 Hz is not included yet ...

// Copyright (c) Sandeep Mistry. All rights reserved. modified by Bo-Erik Sandholm
// Licensed under the MIT license. See LICENSE file in the project root for full license information.

// TimerOne library: https://code.google.com/p/arduino-timerone/
// #include <TimerOne.h>
// DHT library: https://github.com/adafruit/DHT-sensor-library

#include <SPI.h>
#include <BLEPeripheral.h>
#include <BNO055.h>
#include <Wire.h>

// define pins (varies per shield/board) - är dessa lediga?
#define BLE_REQ   0
#define BLE_RDY   1
#define BLE_RST   9

#define A 0X28  // I2C address selection pin LOW
#define B 0x29  //           - " -           HIGH

BNO055 mySensor(A);

BLEPeripheral blePeripheral = BLEPeripheral(BLE_REQ, BLE_RDY, BLE_RST);

BLEService bno055Service = BLEService("CCC0");
BLEFloatCharacteristic Wbno055Characteristic = BLEFloatCharacteristic("CCC1", BLERead | BLENotify);
BLEFloatCharacteristic Xbno055Characteristic = BLEFloatCharacteristic("CCC1", BLERead | BLENotify);
BLEFloatCharacteristic Ybno055Characteristic = BLEFloatCharacteristic("CCC2", BLERead | BLENotify);
BLEFloatCharacteristic Zbno055Characteristic = BLEFloatCharacteristic("CCC3", BLERead | BLENotify);
BLEDescriptor Wbno055Descriptor = BLEDescriptor("2901", "Wquaternion");
BLEDescriptor Xbno055Descriptor = BLEDescriptor("2901", "Xquaternion");
BLEDescriptor Ybno055Descriptor = BLEDescriptor("2901", "Yquaternion");
BLEDescriptor Zbno055Descriptor = BLEDescriptor("2901", "Zquaternion");

volatile bool readFromSensor = false;

float lastTempReading;
float lastHumidityReading;

void setup() {
  mySensor.init();
  Serial.begin(115200);
#if defined (__AVR_ATmega32U4__)
  delay(5000);  //5 seconds delay for enabling to see the start up comments on the serial board
#endif

  blePeripheral.setLocalName("Temperature");

  blePeripheral.setAdvertisedServiceUuid(bno055Service.uuid());
  blePeripheral.addAttribute(bno055Service);
  blePeripheral.addAttribute(Wbno055Characteristic);
  blePeripheral.addAttribute(Wbno055Descriptor);
  blePeripheral.addAttribute(Xbno055Characteristic);
  blePeripheral.addAttribute(Xbno055Descriptor);
  blePeripheral.addAttribute(Ybno055Characteristic);
  blePeripheral.addAttribute(Ybno055Descriptor);
  blePeripheral.addAttribute(Zbno055Characteristic);
  blePeripheral.addAttribute(Zbno055Descriptor);

  blePeripheral.setEventHandler(BLEConnected, blePeripheralConnectHandler);
  blePeripheral.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);

  blePeripheral.begin();

  Serial.println(F("BLE Temperature Sensor Peripheral"));
}

void loop() {
  blePeripheral.poll();

  if (readFromSensor) {
    setbno055DescriptorValue();
   // setHumidityCharacteristicValue();
    readFromSensor = false;
  }
}

void timerHandler() {
  readFromSensor = true;
}

void setbno055DescriptorValue() {
   //Quaternions
     mySensor.readQuat();

  //if (!isnan(reading) && significantChange(lastTempReading, reading, 0.5)) {
    Wbno055Characteristic.setValue(mySensor.quat.q0);
    Xbno055Characteristic.setValue(mySensor.quat.q1);
    Ybno055Characteristic.setValue(mySensor.quat.q2);
    Zbno055Characteristic.setValue(mySensor.quat.q3);
    
    Serial.print(F("Wquaternion: "));  Serial.println(mySensor.quat.q0);

    // lastTempReading = reading;
 // }
}

boolean significantChange(float val1, float val2, float threshold) {
  return (abs(val1 - val2) >= threshold);
}

void blePeripheralConnectHandler(BLECentral& central) {
  Serial.print(F("Connected event, central: "));
  Serial.println(central.address());
}

void blePeripheralDisconnectHandler(BLECentral& central) {
  Serial.print(F("Disconnected event, central: "));
  Serial.println(central.address());
}

sandeepmistry, 2017-04-17:
See here for a list of supported typed characteristics: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/API.md#typed-blecharacteristics

See here for how to setup a bond store for pairing: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/API.md#bond-store - note all characteristics will be private if this is setup and no pairing pin is support - only just works.

closed 2017-04-17 by sandeepmistry