sandeepmistry/arduino-BLEPeripheral #144: Ibeacon example not transmitting Major, Minor, Txpower

vanbwodonk, 2017-04-19:
I try ibeacon with nrf52 example, ibeacon scanned with missing major, minor and txpower.
Then i check data transmitting with nRF connect on android. RAW data showed only uuid. This RAW data i record from nRF connect.

`0x02010615FF4C000215A196C876DE8C4C47AB5AD7AFD5AE7127 `

Last 16 byte must from uuid. 
-Arduino -
`char* uuidString             = "a196c876-de8c-4c47-ab5a-d7afd5ae7127";`
-RAWdata-
`A196C876DE8C4C47AB5AD7AFD5AE7127`

Any suggestion?

sandeepmistry, 2017-04-24:
@vanbwodonk could you please try the change from PR #148 and provide your feedback.

vanbwodonk, 2017-04-25:
@sandeepmistry it's worked. Major, minor, TXpower now transmitted. Thanks !!

sandeepmistry, 2017-04-25:
@vanbwodonk great, thanks for trying it out. I've merged #148.

PR #148 (2017-04-25): Fix length issues with nRF52
Should resolve #144.

cc/ @vanbwodonk
closed 2017-04-25 by sandeepmistry