sandeepmistry/arduino-BLEPeripheral #146: begin() problem?

panzoide, 2017-04-21:
i have an olimexino-nano-ble, and i try everithing that i could, but i couldnt even get the starter.ino working.

it stopjust after i call "blePeripheral.begin();"

i did make it work with the "olimex libraries", so i know that the pins for REQN, RDYN and reset are correct, 

What could be the problem?

thx

sandeepmistry, 2017-04-23:
@panzoide you'll have to pass in the correct pins into the `BLEPeripheral` constructor. Please see: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/API.md#constructor

What Arduino board are you using with the `olimexino-nano-ble`?

panzoide, 2017-04-23:
im using an olimexino-nano,
and i did pass the correct pins to the BLEperipheral, they are REQ=6 RDY=4 RST=5, they work for this library, but doesnt have eddystone (https://github.com/NordicSemiconductor/ble-sdk-arduino)

sandeepmistry, 2017-04-23:
@panzoide could you please try uncommenting the following define: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/src/nRF8001.cpp#L6

To enable debug output.

panzoide, 2017-04-24:
it just stay at: 

`Evt Device Started: Setup
07 06 00 00 03 02 41 FE 
`
it's the only output after 3 minutes running

sandeepmistry, 2017-04-24:
Ok, it's a bit difficult for me to figure this out without a physical board, but just to confirm you have `BLEPeripheral(6, 4, 5)`?

What version of the Arduino IDE are you using?

panzoide, 2017-04-24:
i have the exact starter.ino with 
`BLEPeripheral blePeripheral = BLEPeripheral(6,4,5);`

and im using the 1.8.1v

sandeepmistry, 2017-05-07:
@panzoide maybe this is an SPI speed issue.

Could you try changing: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/f591da1a841bf6b0a5c1eac4d1255ba6e71154e8/src/nRF8001.cpp#L133 ? Maybe DIV16 will be better?

CTumminaro, 2017-05-10:
I am new to Arduino. Using the Arduino Uno R3 with a Adafruit Bluefruit LE nRF8001. I'm following this tutorial: https://library.oreilly.com/book/0636920031932/make-bluetooth/22.xhtml?ref=toc#connecting-to-an-arduino I have the correct pins passing through to BLEPeripheral per the tutorial. I have also tried downgrading to a few versions of this library with no luck.

I am having the same issue as @panzoide - I tried changing that line to DIV16 with no luck.

CTumminaro, 2017-05-21:
It appears the problem I was having was the example cheating a bit  on the uuid:

BLEService lightswitch = BLEService("FF10");

Changing this to 32 character uuid clears up the begin() issue that I was having.

dots123, 2018-06-02:
I've figured out my problem: I don't have a low frequency crystal. Only 32Mhz crystal. And every time when I make a short between P0.00 and P0.01 - LFCLK pins, the module appears in the smartphone app, but I can't connect because of the short. 
I've compiled with internal oscillator, by selecting from Tools>RC oscillator and it works, I can connect to it.
Now I try to figured out how to disable the external LFCLK on Arduino Primo board, cause it doesn't have a menu in Tools dedicated to it.

mayhd3, 2021-02-06:
> it just stay at:
> 
> `Evt Device Started: Setup 07 06 00 00 03 02 41 FE `
> it's the only output after 3 minutes running

Sorry to dig this back up, having this problem in 2021 with the same board. Did you ever figure it out?

mayhd3, 2021-02-07:
Some debugging has shown that it is actually freezing in `this->sendSetupMessage(&setupMsg);` from line 314 of nrf8001.cpp, I will investigate more in there...

mayhd3, 2021-02-07:
...because the loop on line 1504 never terminates. I am not sure where to go from here.

mayhd3, 2021-02-07:
Could it be working correctly? I do not understand what the output after `Evt Device Started: Setup` is, I see eight bytes while the bdaddr should be six right?