sandeepmistry/arduino-BLEPeripheral #149: Trying to read from the ANCS Data Source

jacobmh1177, 2017-05-04:
Hello @claudioarduino and I are working on getting notification attributes from ANCS. Right now we know that we are formatting the command buffer sent to the ANCS control source correctly but we can't read anything from the data source. I saw from an earlier issue this may be due to the fact that we are subscribing to two BLENotify characterisitics. When I subscribe to the Notification Source first I can read all the notification meta data, but when I subscribe to the Data Source first I don't get any information from the Notification Source. How can we get around this issue?

Thanks :)

sandeepmistry, 2017-05-08:
@jacobmh1177 do you have a sketch to reproduce this? what debugging have you done so far?

claudioarduino, 2017-05-09:
> Data: 04/05/2017 23.04
>
>
>
> Ogg: [sandeepmistry/arduino-BLEPeripheral] Trying to read from the ANCS Data Source (#149)

Hi @jacobmh1177 I try to introduce the subscribe request before the ControlPoint command in Notification function. It works properly, I think that somebody did not give us the response, because this is very simple question for expert people. Use the solution as you want, attach it in the forum if you prefer.At this point I'm working for Android too, I think to use ASN but there are very few examples and they don't work properly, have you a suggestion? Do you know if there are some examples in the web?Thank you and B.R. 

----Messaggio originale----

sandeepmistry, 2018-01-08:
Closing for now due to lack of feedback.

closed 2018-01-08 by sandeepmistry