sandeepmistry/arduino-BLEPeripheral #152: setConnectionInterval() nRF52

stevensam2013, 2017-05-11:
I have been trying to use the setConnectionInterval() method to set the interval range to between 7.5 and 15ms, but setting the min and max to 6 and 12 respectively, but I can't seem to get the central device (Android phone) to report back anything other than the default.

I have tried a range of different values for the min and max, but never seem to get any change at all.

I have tried calling setConnectionInterval() before connecting, after connection, within the connection callback, outside of the callback, but nothing seems to make a difference. Am I missing a step, or is this an issue with the nRF52?

stevensam2013, 2017-05-19:
Calling the setConnectionInterval() method seems to ultimately call 'sd_ble_gap_conn_param_update', which is returning NRF_SUCCESS, when it is called, but when I check the Preferred Connection Parametrs through the nRF Connect App, they have not changed from the default.

Looking into the SDK documentation, it seems that for nRF52, the function that should be called is 'ble_conn_params_change_conn_params', however if I try to use this, then I get the error saying.
'ble_conn_params_change_conn_params' was not declared in this scope

Any clarification around this would be appreciated, as I am unable to get the Connection Parameters to update.

stevensam2013, 2017-06-09:
In lines 179-182 of nRF51822.cpp, the connection intervals are hard coded, and the requested connection interval parameters always show up on the central as the values that are set in this file.

dsanders11, 2017-09-10:
@stevensam2013, did you ever resolve this? You need to call `setConnectionInterval` before `begin`, in my experience.

sandeepmistry, 2018-01-08:
Closing for now due to lack of feedback.

closed 2018-01-08 by sandeepmistry