sandeepmistry/arduino-BLEPeripheral #153: Starter example connection problem

claudioarduino, 2017-05-17:
Hi everyone, I'm trying to work with the "starter example", but it's not possible to have a stable connection between the device and the BLE. Here's the response I have from the Serial monitor when I try to pair to the device:

`BLE` local-name
Connected to central: 7c:c4:d2:17:cd:da
Disconnected from central: `7c:c4:d2:17:cd:da`

I'm using Android 6.0.1 with Arduino mega 2560 and nrf8001 from Adafruit.
Thanks

claudioarduino, 2017-05-19:
I tried to enable nrf8001 debug and this is what i see from the serial monitor :

Connect to serial portCOM4 at 115200
Evt Device Started: Setup 
07 06 00 00 03 02 41 FE 
Evt Cmd Rsp: Transaction Continue
1F 06 10 00 00 00 00 00 00 00 04 00 05 01 01 00 00 06 00 06 00 00 00 00 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
1F 06 10 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 50 00 03 90 00 FF 
Evt Cmd Rsp: Transaction Continue
1F 06 10 38 FF FF 02 58 0A 05 00 00 40 00 00 00 40 00 00 00 50 00 00 00 40 00 01 02 01 02 01 02 
Evt Cmd Rsp: Transaction Continue
05 06 10 54 00 02 
Evt Cmd Rsp: Transaction Continue
19 06 70 00 02 02 F0 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
19 06 70 16 09 0A 6C 6F 63 61 6C 2D 6E 61 6D 65 00 00 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
0E 06 20 00 04 04 02 02 00 01 28 00 01 00 18 
Evt Cmd Rsp: Transaction Continue
11 06 20 0B 04 04 05 05 00 02 28 03 01 02 03 00 00 2A 
Evt Cmd Rsp: Transaction Continue
0C 06 20 19 04 04 13 07 00 03 2A 00 01 
Evt Cmd Rsp: Transaction Continue
16 06 20 22 41 72 64 75 69 6E 6F 00 00 04 00 00 00 00 00 02 00 00 05 
Evt Cmd Rsp: Transaction Continue
11 06 20 35 04 04 05 05 00 04 28 03 01 02 05 00 01 2A 
Evt Cmd Rsp: Transaction Continue
0C 06 20 43 04 04 02 02 00 05 2A 01 01 
Evt Cmd Rsp: Transaction Continue
05 06 20 4C 00 00 
Evt Cmd Rsp: Transaction Continue
0E 06 20 4E 04 04 02 02 00 06 28 00 01 01 18 
Evt Cmd Rsp: Transaction Continue
11 06 20 59 04 04 05 05 00 07 28 03 01 20 08 00 05 2A 
Evt Cmd Rsp: Transaction Continue
0C 06 20 67 24 00 04 00 00 08 2A 05 01 
Evt Cmd Rsp: Transaction Continue
07 06 20 70 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
0E 06 20 74 46 14 03 02 00 09 29 02 01 00 00 
Evt Cmd Rsp: Transaction Continue
0E 06 20 7F 04 04 02 02 00 0A 28 00 01 F0 FF 
Evt Cmd Rsp: Transaction Continue
11 06 20 8A 04 04 05 05 00 0B 28 03 01 0A 0C 00 F1 FF 
Evt Cmd Rsp: Transaction Continue
0C 06 20 98 46 14 02 01 00 0C FF F1 01 
Evt Cmd Rsp: Transaction Continue
04 06 20 A1 00 
Evt Cmd Rsp: Transaction Continue
0C 06 20 A2 04 04 05 05 00 0D 29 01 01 
Evt Cmd Rsp: Transaction Continue
08 06 20 AB 76 61 6C 75 65 
Evt Cmd Rsp: Transaction Continue
04 06 20 B0 00 
Evt Cmd Rsp: Transaction Continue
0D 06 40 00 00 00 01 00 80 04 00 03 00 00 
Evt Cmd Rsp: Transaction Continue
0D 06 40 0A 00 00 01 00 80 04 00 05 00 00 
Evt Cmd Rsp: Transaction Continue
0D 06 40 14 00 00 01 00 04 04 00 08 00 09 
Evt Cmd Rsp: Transaction Continue
0D 06 40 1E 00 00 01 00 90 04 00 0C 00 00 
Evt Cmd Rsp: Transaction Continue
06 06 F0 00 03 DB 0A 
Evt Cmd Rsp: Transaction Complete
BLE-Starter
Evt Device Started: Standby
Advertising started.
Device address = f4:d9:04:58:8e:8c
Device address type = 2
Peripheral address: f4:d9:04:58:8e:8c
Evt Connected 5f:dc:e2:b5:aa:04
Peripheral connected to central: 5f:dc:e2:b5:aa:04
Connected to central: 5f:dc:e2:b5:aa:04
Evt Pipe Status 
37
0
Evt Disconnected/Advertising timed out
Peripheral disconnected from central: 5f:dc:e2:b5:aa:04
Advertising started.
Disconnected from central: 5f:dc:e2:b5:aa:04

sandeepmistry, 2017-06-18:
To support pairing, you need to set a`BLEBondStore`, a few of the examples illustrate how to do this.

closed 2017-06-18 by sandeepmistry