sandeepmistry/arduino-BLEPeripheral #154: Redbear BLE Nano v1.5 can't write to remote characteristic

jacobmh1177, 2017-05-19:
I'm trying to implement the ancs example again, except this time using a Redbear BLE Nano. I can subscribe to the Notification Source just fine. However ancsControlPointCharacteristic.canWrite() returns False and ancsControlPointCharacteristic.write(...) also returns false. Have you encountered this error before on this board where you can't write to remote characteristics? Here is the code snippet below:

unsigned long uid = notification.notificationUid;
  byte buffer[8];
  buffer[0] = 0;
  buffer[4] = (uid >> 24) & 0xFF;
  buffer[3] = (uid >> 16) & 0xFF;
  buffer[2] = (uid >> 8) & 0xFF;
  buffer[1] = uid & 0xFF;
  buffer[5] = 3;//ANCS_NOTIFICATION_ATTRIBUTE_MESSAGE;
  buffer[6] = 20; //ANCS_NOTIFICATION_ATTRIBUTE_MESSAGE;
  buffer[7] = 0; //ANCS_NOTIFICATION_ATTRIBUTE_MESSAGE;
  BLEUtil::printBuffer(buffer, 8);
  Serial.println("Sending notification attribute buffer ");
  Serial.println();
  Serial.println(ancsControlPointCharacteristic.canWrite());
  Serial.println(ancsControlPointCharacteristic.write((const unsigned char*)buffer, 8));

claudioarduino, 2017-05-19:
@jacobmh1177  Hi jacob, I commented you on issue #142 . I solved the problem with the data source. Did you see?

jacobmh1177, 2017-05-19:
@claudioarduino Yes, and it works great thank you so much :) I switched boards to the redbear ble nano because of memory issues. I'm trying to get our code working on the ble nano and created this issue because it won't even write to the control point (as opposed to our original problem using the nrf8001 where we could write to the control point but were having trouble subscribing to the data source). If you've got any tips here I'd love your help again!