sandeepmistry/arduino-BLEPeripheral #159: HID + ANCS

claudioarduino, 2017-05-26:
@sandeepmistry  Hi Sandeep, I'm trying to release a unique interface with both HID and ANCS examples. I made a merge of the two examples, but the two services doesn't work simultaneously, it seems that it works only the hid service. How can i do to make them work together? Here's the code I wrote:

```
#include "Arduino.h"
#include <SPI.h>
#include <BLEHIDPeripheral.h>
#include <BLEKeyboard.h>
#include <BLEMultimedia.h>

#include <BLEPeripheral.h>
#include <BLEUtil.h>

// define pins (varies per shield/board)
#define BLE_REQ   10
#define BLE_RDY   2
#define BLE_RST   9

unsigned char flagHome = 0;
unsigned char v = 0;
bool deviceConnected = false;
bool deviceCentral = false;
bool disableNotify = false;

//#define ANDROID_CENTRAL

// create peripheral instance, see pinouts above

BLEHIDPeripheral bleHID = BLEHIDPeripheral(BLE_REQ, BLE_RDY, BLE_RST);
BLEBondStore                     bleBondStore;

BLERemoteService                 ancsService                              = BLERemoteService("7905f431b5ce4e99a40f4b1e122d00d0");

// remote characteristics
BLERemoteCharacteristic          ancsNotificationSourceCharacteristic     = BLERemoteCharacteristic("9fbf120d630142d98c5825e699a21dbd", BLENotify);
BLERemoteCharacteristic          ancsControlPointCharacteristic           = BLERemoteCharacteristic("69d1d8f345e149a898219bbdfdaad9d9", BLEWrite);
BLERemoteCharacteristic          ancsDataSourceCharacteristic             = BLERemoteCharacteristic("22eac6e924d64bb5be44b36ace7c7bfb", BLENotify);

BLEKeyboard bleKeyboard;
BLEMultimedia bleMultimedia;

//Notification
byte ANCS_COMMAND_ID_GET_NOTIF_ATTRIBUTES                      = 0;
byte ANCS_COMMAND_ID_PERFORM_NOTIF_ACTION                   = 2;
byte ANCS_NOTIFICATION_ATTRIBUTE_APP_IDENTIFIER                 = 0;
byte ANCS_NOTIFICATION_ATTRIBUTE_TITLE                                   = 1;
byte ANCS_NOTIFICATION_ATTRIBUTE_SUBTITLE                             = 2;
byte ANCS_NOTIFICATION_ATTRIBUTE_MESSAGE                            = 3;
byte ANCS_NOTIFICATION_ATTRIBUTE_MESSAGE_SIZE                    = 4;
byte ANCS_NOTIFICATION_ATTRIBUTE_DATE                                    = 5;
byte ANCS_NOTIFICATION_ATTRIBUTE_POSITIVE_ACTION_LABEL    = 6;
byte ANCS_NOTIFICATION_ATTRIBUTE_NEGATIVE_ACTION_LABEL  = 7;
byte ANCS_ACTION_ID_POSITIVE                                                       = 0;
byte ANCS_ACTION_ID_NEGATIVE                                                      = 1;
//Notification

char risp = '0';
long categ;

void setup() {
    Serial.begin(9600);
  #if defined (__AVR_ATmega32U4__)
    while(!Serial);
  #endif

  #ifdef ANDROID_CENTRAL
       bleHID.setReportIdOffset(1);
  #endif

      bleBondStore.clearData();

[... 451 lines cut ...]
0 
Evt Bond Status
0 
Peripheral bonded: 71:b0:7e:40:c8:4b
Remote bonded event, central: 71:b0:7e:40:c8:4b
Evt Pipe Status 
57F 
2000
Peripheral discovered central remote services: 71:b0:7e:40:c8:4b
Remote services discovered event, central: 71:b0:7e:40:c8:4b 
Subscribed to Data Source 
Evt Pipe Status 
257F 
0
Evt Pipe Status 
25FF
0 
Evt Pipe Status 
27FF 
0
Timing change received conn Interval: 0xC
```
Do you have any hint on how I can make the two services work together?
Thanks for your help.
Claudio

sandeepmistry, 2017-05-29:
I think it's probably related to the HID service UUID not being advertised.

claudioarduino, 2017-05-29:
@sandeepmistry Hi sandeep, I tried to add `bleHID.setAdvertisedServiceUuid("1812");` but it seems nothing changes.
In the debug I can see that it subscribes to Data Source but not to Notification Source.  I still have the same error : ACI Evt Pipe Error: Pipe #:13  Pipe Error Code: 0x92. What it means?
Thanks,
Claudio

claudioarduino, 2017-05-30:
@sandeepmistry  Hi sandeep, I tried to create two different `void setup`, one for the ancs service and the other one for the hid service. When I load only ancs service or only hid service it works properly. Now I want to use both hid and ancs, but when I skip from ancs service (the first setup) to hid service (the second setup, calling `void setup2();` inside the main `loop`) the device disconnect and I'm not able to connect again. I'd like to remain connected.
This is the code I released:

void setup() {
    Serial.begin(9600);
  #if defined (__AVR_ATmega32U4__)
    while(!Serial);
  #endif

  #ifdef ANDROID_CENTRAL
       bleHID.setReportIdOffset(1);
  #endif

      bleBondStore.clearData();

      bleHID.setBondStore(bleBondStore);

      bleHID.setServiceSolicitationUuid(ancsService.uuid());
      bleHID.setLocalName("HID ANCS");
      bleHID.setDeviceName("BLE HID ANCS");
      bleHID.setAppearance(0x0080);

      bleHID.addRemoteAttribute(ancsService);
      bleHID.addRemoteAttribute(ancsNotificationSourceCharacteristic);
      bleHID.addRemoteAttribute(ancsControlPointCharacteristic);
      bleHID.addRemoteAttribute(ancsDataSourceCharacteristic);

 //     bleHID.setAdvertisedServiceUuid("1812");
 //     bleHID.addHID(bleKeyboard);
 //     bleHID.addHID(bleMultimedia);

      bleHID.setEventHandler(BLEConnected, blePeripheralConnectHandler);
      bleHID.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);
      bleHID.setEventHandler(BLEBonded, blePeripheralBondedHandler);
      bleHID.setEventHandler(BLERemoteServicesDiscovered, blePeripheralRemoteServicesDiscoveredHandler);

      ancsNotificationSourceCharacteristic.setEventHandler(BLEValueUpdated, ancsNotificationSourceCharacteristicValueUpdated);
      ancsDataSourceCharacteristic.setEventHandler(BLEValueUpdated, ancsDataSourceCharacteristicCharacteristicValueUpdated);

      bleHID.begin();

    Serial.println(F("BLE HID ANCS"));

}

void setup2() {
	setup2_flag = true;
    bleBondStore.clearData();

	bleHID.setBondStore(bleBondStore);
    bleHID.setAdvertisedServiceUuid("1812");

 //   bleHID.setLocalName("HID ANCS");
 //   bleHID.setDeviceName("BLE HID ANCS");
 //   bleHID.setAppearance(0x0080);
    bleHID.addHID(bleKeyboard);
    bleHID.addHID(bleMultimedia);

    bleHID.setEventHandler(BLEConnected, blePeripheralConnectHandler);
    bleHID.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);
    bleHID.setEventHandler(BLEBonded, blePeripheralBondedHandler);
    bleHID.begin2();
    Serial.println(F("BLE HID ANCS 2"));
}

The debug when I call the `void setup2()` is:

>>Send to COM4: "q"<<
BLE HID ANCS 2
Evt Device Started: Setup
07 06 00 00 03 02 41 FE 
Evt Cmd Rsp: Transaction Continue
1F 06 10 00 00 00 00 00 02 00 05 00 07 01 01 00 00 06 00 06 00 00 00 00 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
1F 06 10 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 50 00 03 90 00 FF 
Evt Cmd Rsp: Transaction Continue
1F 06 10 38 FF FF 02 58 0A 05 00 00 40 00 00 00 40 00 00 00 50 00 00 00 40 00 01 02 01 02 01 02 
Evt Cmd Rsp: Transaction Continue
05 06 10 54 01 02 
Evt Cmd Rsp: Transaction Continue 
19 06 70 00 15 10 D0 00 2D 12 1E 4B 0F A4 99 4E CE B5 31 F4 05 79 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
19 06 70 16 09 08 48 49 44 20 41 4E 43 53 00 00 00 00 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
0E 06 20 00 04 04 02 02 00 01 28 00 01 12 18 
Evt Cmd Rsp: Transaction Continue
11 06 20 0B 04 04 05 05 00 02 28 03 01 02 03 00 4A 2A 
Evt Cmd Rsp: Transaction Continue
0C 06 20 19 06 0C 05 04 00 03 2A 4A 01  
Evt Cmd Rsp: Transaction Continue
07 06 20 22 11 01 00 03 
Evt Cmd Rsp: Transaction Continue
11 06 20 26 04 04 05 05 00 04 28 03 01 04 05 00 4C 2A 
Evt Cmd Rsp: Transaction Continue
0C 06 20 34 46 18 02 01 00 05 2A 4C 01  
Evt Cmd Rsp: Transaction Continue
04 06 20 3D 00  
Evt Cmd Rsp: Transaction Continue 
11 06 20 3E 04 04 05 05 00 06 28 03 01 02 07 00 4B 2A 
Evt Cmd Rsp: Transaction Continue
0C 06 20 4C 06 0C 4B 4A 00 07 2A 4B 01 
Evt Cmd Rsp: Transaction Continue
1F 06 20 55 05 01 09 06 A1 01 85 00 75 01 95 08 05 07 19 E0 29 E7 15 00 25 01 81 02 95 01 75 08 
Evt Cmd Rsp: Transaction Continue
1F 06 20 71 81 03 95 05 75 08 15 00 26 A4 00 05 07 19 00 2A A4 00 81 00 C0 05 0C 09 01 A1 01 85 
Evt Cmd Rsp: Transaction Continue 
15 06 20 8D 01 19 00 2A 9C 02 15 00 26 9C 02 95 01 75 10 81 00 C0 
Evt Cmd Rsp: Transaction Continue
11 06 20 9F 04 04 05 05 00 08 28 03 01 12 09 00 4D 2A 
Evt Cmd Rsp: Transaction Continue
0C 06 20 AD 14 0C 08 00 00 09 2A 4D 01 
Evt Cmd Rsp: Transaction Continue
0B 06 20 B6 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
0E 06 20 BE 46 14 03 02 00 0A 29 02 01 00 00 
Evt Cmd Rsp: Transaction Continue 
0C 06 20 C9 04 04 02 02 00 0B 29 08 01 
Evt Cmd Rsp: Transaction Continue
05 06 20 D2 00 01 
Evt Cmd Rsp: Transaction Continue
11 06 20 D4 04 04 05 05 00 0C 28 03 01 12 0D 00 4D 2A 
Evt Cmd Rsp: Transaction Continue
0C 06 20 E2 14 0C 02 00 00 0D 2A 4D 01  
Evt Cmd Rsp: Transaction Continue
05 06 20 EB 00 00 
Evt Cmd Rsp: Transaction Continue
0E 06 20 ED 46 14 03 02 00 0E 29 02 01 00 00 
Evt Cmd Rsp: Transaction Continue
0C 06 20 F8 04 04 02 02 00 0F 29 08 01  
Evt Cmd Rsp: Transaction Continue 
05 06 21 01 01 01  
Evt Cmd Rsp: Transaction Continue 
04 06 21 03 00 
Evt Cmd Rsp: Transaction Continue 
0D 06 40 00 00 00 01 00 80 04 00 03 00 00 
Evt Cmd Rsp: Transaction Continue
0D 06 40 0A 00 00 01 00 08 04 00 05 00 00 
Evt Cmd Rsp: Transaction Continue
0D 06 40 14 00 00 01 00 80 04 00 07 00 00 
Evt Cmd Rsp: Transaction Continue
0D 06 40 1E 00 00 01 00 82 04 00 09 00 0A 
Evt Cmd Rsp: Transaction Continue
0D 06 40 28 00 00 01 00 82 04 00 0D 00 0E 
Evt Cmd Rsp: Transaction Continue
06 06 F0 00 03 51 3A

As you can see it's incomplete and the device disconnect.
Do you know where I'm wrong?
Thanks,
Claudio

sandeepmistry, 2017-06-18:
It could be the board is running out of RAM?

claudioarduino, 2017-06-19:
@sandeepmistry Hi sandeep, I tried to reduce the sketch and see the amount of RAM used.
The first test, that I did, is to add the HID Multimedia(`bleHID.addHID(bleMultimedia)`) and the ANCS service; the multimedia, the control point and the data source work properly, the notification source doesn't work. 
Below the output of the compilation:

'Building target: HID_ANCS'
'Printing size:'
"C:\Users\ab007567\eclipse\arduinoPlugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-size" -A "C:/Users/ab007567/HID_Arduino_1/HID_ANCS/Release/HID_ANCS.elf"
C:/Users/ab007567/HID_Arduino_1/HID_ANCS/Release/HID_ANCS.elf  :
section                     size      addr
.data                        720   8389120
.text                      22208         0
.bss                         996   8389840
.comment                      17         0
.note.gnu.avr.deviceinfo      64         0
.debug_info                 3004         0
.debug_abbrev               2842         0
.debug_line                   26         0
.debug_str                   998         0
Total                      30875

'Finished building target: HID_ANCS'

In the second test I add only the ancs service, without the HID Multimedia(`//bleHID.addHID(bleMultimedia)`), and it works.
Below the output of the complilation(as you can see it's similar):

'Building target: HID_ANCS'
'Printing size:'
"C:\Users\ab007567\eclipse\arduinoPlugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-size" -A "C:/Users/ab007567/HID_Arduino_1/HID_ANCS/Release/HID_ANCS.elf"
C:/Users/ab007567/HID_Arduino_1/HID_ANCS/Release/HID_ANCS.elf  :
section                     size      addr
.data                        720   8389120
.text                      22114         0
.bss                         996   8389840
.comment                      17         0
.note.gnu.avr.deviceinfo      64         0
.debug_info                 3004         0
.debug_abbrev               2842         0
.debug_line                   26         0
.debug_str                   998         0
Total                      30781

'Finished building target: HID_ANCS'

If you could help me thank you very much.
Claudio