sandeepmistry/arduino-BLEPeripheral #160: Windows 10 Driver Error

mholin1, 2017-06-08:
Hello.  trying to make a Bluetooth mouse/keyboard for a Windows 10 PC and starting with the "HID_test" Example from the Arduino BLEPeripheral Library on a Sparkfun nRF52832 Breakout Board using only the mouse portion of HID_test for now.  The PC sees "HID" local name and starts to pair but then after few seconds displays "Driver Error" and drops the connection.  A commercial Bluetooth mouse pairs and works with the same PC however.

Also found the function:

bleHID.clearBondStoreData();

from HID_test conflicts with the nRF52832 Breakout Board bootloader so just using a J-Link to program the application for now.

Hoping someone might have some insight into that Driver Error ?

sandeepmistry, 2017-06-18:
@mholin1 I haven't tried this setup myself, but I recall the Windows pairing process being a bit weird compared to Android and iOS.

mholin1, 2017-06-18:
Thanks for taking a look.  Someone on the Nordic Developer Zone was getting a similar "Driver Error" and the resolution seemed to be something about bonding?  Admittedly I am not familiar enough with the BLE spec. or the Library to know how to implement bonding.

Also, can you say if that bleHID.clearBondStoreData(); statement is necessary for pairing?  And any idea why it is conflicting with the bootloader?  It looks like you are putting the bond store up at the last page of flash memory which should be fine as far as I can tell.

mholin1, 2017-07-10:
I moved the bond store to just under the bootloader (from default 0x7F000 to 0x79000) but still getting "Driver Error" messages from Windows 10 while attempting to pair.  Has anybody ever used this Library successfully with a Windows 10 HID application (mouse or keyboard)?

mholin1, 2017-07-27:
Hi Sandeep,

Can you suggest any way I can get help with this?  I set up Nordic's BLE Sniffer and had the Nordic guys take a look but it is not their code and so they really could not help me.  Is there anyone who has used the Library on a Windows 10 HID application?

Thanks,
Mark

--------------------------------------------

 

 
 it on GitHub, or mute
 the thread.

varunhr, 2017-12-01:
Same problem with me as well..... Would some one help us with this issue?

Regards
Varun

mholin1, 2017-12-01:
Hello Varun,  I eventually had to give up.  No real support from any entity involved.  Tried the Cypress PSoC 4 parts and it works fine.  PSoC Creator light years simpler development environment.  Cypress tech support is as abysmal as this Nordic platform but at least the parts work as advertised.

Note:  At least Nordic Tech Support tried to help but it was not their code.

RicoKirstein, 2017-12-08:
I had a problem with the GATT HID driver in windows 10 as well. Uncommenting the line
"bleHIDPeripheral.setReportIdOffset(1);" solved it for me.

mholin1, 2017-12-08:
Thanks for your input.  I don't see that line in the original Example HID_test.ino.  Is it in one of the header files?  Also i do recall there was a conflict between the bootloader so moving the bond store offset to 0x79000 solved that problem but I still got the Win 10 Driver Error.  

So does your device (mouse in my case) now work on Win 10?  If so could you please attach your sketch?

Thanks for posting.

Mark

--------------------------------------------

RicoKirstein, 2017-12-08:
I have my BLE nano 2 on a swd probe, so I haven't ran into any bootloader issued.
The line was in the HID_Keyboard example. I haven't had any successful inputs send from the device yet, but the driver error went away and the device is loaded correctly according to windows. I'm currently on the windows 10 1709 thing. My sketch is just the HID_Keyboard example with the ifdef surrounding bleHIDPeripheral.setReportIdOffset(1) removed.

mholin1, 2017-12-08:
Thanks.  That ifdef isn't even in HID_test.  If you are having issues with Win 10 versions, I do recall one of the recent Microsoft "upgrades" no longer reports the Driver Error but the mouse still does not function.  A commercial BLE mouse does function on the same PC however.

Good Luck.  Please post if you find anything further.

--------------------------------------------

varunhr, 2017-12-11:
Hai thank you very much for the info. I too moved to TI's CC2541 chip(HM 10
module). Its working in win 10 with some bugs. Trying to debug them.

Regards
Varun h r

RicoKirstein, 2017-12-11:
Can you post a link to the libraries you're using? Maybe it's just a problem with the descriptor used in this library.
Cheers
Rico

mholin1, 2017-12-11:
As I recall it is just the Arduino BLEPeripheral Library downloaded straight from the Arduino IDE.  Where you referring to a different Library?

Thanks for the help :)

--------------------------------------------

RicoKirstein, 2017-12-11:
It seems we're missing 2 BLE reports. The standart says there should be 2 more for keyboards at least. 
BootKeyboardOutputReport and BootKeyboardInputReport
For Mouse it should be BootMouseInputReport.
Can you check with a BLE explorer tool if your working devices have those services?
Cheers
Rico

mholin1, 2017-12-11:
Interesting.  You mean the Nordic sniffer on the commercial BLE mouse ?

--------------------------------------------

RicoKirstein, 2017-12-11:
Yes

RicoKirstein, 2018-01-03:
I also finally gave up on this. The nordic sdk keyboard sample works with some changes. I got my usb to ble keyboard adapter working using a max3421E and a blenano2.

mholin1, 2018-01-03:
Glad to hear you got something to work.  I could have probably gotten the Nordic ble mouse example to work but then there would be no code memory room left for the rest of my application due to the 32K limit of the free Keil compiler.  Tried un-checking the various added code options but that just produced tons of compiler errors that would have to track down.  On the other hand, the Cypress hardware and software worked right out of the box.

Just really unfortunate the BLEPeripheral Library does not support Windows 10.

--------------------------------------------

kheek, 2018-04-16:
Hi @sandeepmistry  is it correct the BLEPeripheral Library does not support Windows 10? I tried for many hours to get the HID example working, bonding with Windows 10, but without result.  The sample works fine with Macbook & iPhone, but not with Windows 10. I'm really desperate, because I like to control my powerpoint presentation via a Neoplc UC + NRF52 board. Can you confirm Windows 10 HID is not supported so I can look after alternatives? Greetz from The Netherlands, Klaas Heek

joric, 2018-07-11:
Same here on BLE400 + Core51822 on windows 10. It connects then disconnects and shows "driver error".

andysheen, 2019-05-02:
Sorry to open this, but wondering if anyone has any thoughts on this and have tried getting it working with a BLE HID Gamepad. I've got something working perfectly in OSX, but windows 10 will pair and connect, but display no input. Playing with the offset seems to allow some kind of data to be input appear, but definitely not valid.

Gallium70, 2020-03-30:
Moving bond store actually DOES solve the problem, at least for me...but not with a shift of 6(as mentioned in  #https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/164). 
I tried a shift of 10(i.e. BLEBondStore(10) ), and it works. Evil magic numbers.
And i'm not sure if anyone still cares about this in 2020...hope this might help someone.

Fressco0, 2023-05-25:
> Moving bond store actually DOES solve the problem, at least for me...but not with a shift of 6(as mentioned in ##164). I tried a shift of 10(i.e. BLEBondStore(10) ), and it works. Evil magic numbers. And i'm not sure if anyone still cares about this in 2020...hope this might help someone.

Even in 2023!
Where to change what to do this? And does then also needs to change smth in e.g. clearBondStoreData();?

mholin1, 2023-05-25:
Hey thanks for the update. I don't recall if I tried different offsets. Very painful experience I really do not care to go back to :( Cypress app code worked right out of the box but good to know you had success.

Thx,
M.