sandeepmistry/arduino-BLEPeripheral #161: Error compiling for board Arduino 101

salmanfarisvp, 2017-06-09:
When is run test code in **arduino 101** 
```
Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino 101"

In file included from C:\Users\Faris\OneDrive\Documents\Arduino\libraries\BLEPeripheral\src/utility/lib_aci.h:34:0,

                 from C:\Users\Faris\OneDrive\Documents\Arduino\libraries\BLEPeripheral\src/nRF8001.h:9,

                 from C:\Users\Faris\OneDrive\Documents\Arduino\libraries\BLEPeripheral\src/BLEPeripheral.h:26,

                 from C:\Users\Faris\OneDrive\Documents\Arduino\libraries\BLEPeripheral\examples\test\test.ino:12:

C:\Users\Faris\OneDrive\Documents\Arduino\libraries\BLEPeripheral\src/utility/hal_platform.h:75:6: error: #error "Unsupported platform"

     #error "Unsupported platform"

      ^

Multiple libraries were found for "BLEPeripheral.h"
 Used: C:\Users\Faris\OneDrive\Documents\Arduino\libraries\BLEPeripheral
 Not used: C:\Users\Faris\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\libraries\CurieBLE
exit status 1
Error compiling for board Arduino/Genuino 101.

```

anweiss, 2017-06-16:
Also seeing this issue.

salmanfarisvp, 2017-06-17:
any solutions...?

sandeepmistry, 2017-06-18:
Please see the read me for compatible hardware: https://github.com/sandeepmistry/arduino-BLEPeripheral#compatible-hardware

For the 101, you'll have to use the `CurieBLE` library.

closed 2017-06-18 by sandeepmistry

salmanfarisvp, 2017-06-18:
am using the` CurieBLE` library provided with the` intel 101` board definition , but still getting the error .

neillindberg, 2017-07-26:
@salmanfarisvp 
I had included CurieBLE.h via the menus. When I did that there was a whole stack of includes. I'm guessing you may have done the same?
When I removed all the extras I started seeing no errors in compile time. I've yet to complete my mission and make my garage door controlled from my phone, but I thought I should mention my experience.