sandeepmistry/arduino-BLEPeripheral #163: Redefinition/previous definition of class errors during compilation with PlatformIO

anweiss, 2017-06-16:
Getting a number of `redefinition of class ...` and `previous definition of class ... ` errors when using PlatformIO. I have included the library in to my project via `pio lib install BLEPeripheral`. Not sure why this is occurring because the header files that it's referencing include header guards.

sandeepmistry, 2017-06-18:
@anweiss do you have more context for the compile errors? Including target board etc.

I don't really use PlatformIO my self, maybe @vojtamolda can help you out ...

anweiss, 2017-06-18:
Sure thing. I'm targeting the Arduino 101. Indeed, one can use the CurieBLE library, however, I'm basing my particular use case off of this code -> https://github.com/adafruit/Adafruit_Curie_Bot/tree/master/Ada_CurieBot_RC ... which inherits from this library (https://github.com/adafruit/Adafruit_Curie_Bot/blob/master/Ada_CurieBot_RC/BLESerial.h#L18). 

Some more context below:

```
In file included from .piolibdeps/BLEPeripheral_ID259/src/utility/lib_aci.h:34:0,
from .piolibdeps/BLEPeripheral_ID259/src/nRF8001.h:9,
from .piolibdeps/BLEPeripheral_ID259/src/BLEPeripheral.h:26,
from src/Ada_CurieBot_RC.cpp:23:
.piolibdeps/BLEPeripheral_ID259/src/utility/hal_platform.h:75:6: error: #error "Unsupported platform"
#error "Unsupported platform"
^
In file included from .piolibdeps/BLEPeripheral_ID259/src/utility/lib_aci.h:35:0,
from .piolibdeps/BLEPeripheral_ID259/src/nRF8001.h:9,
from .piolibdeps/BLEPeripheral_ID259/src/BLEPeripheral.h:26,
from src/Ada_CurieBot_RC.cpp:23:
.piolibdeps/BLEPeripheral_ID259/src/utility/hal_aci_tl.h:60:0: warning: "UNUSED" redefined [enabled by default]
#define UNUSED      255
^
In file included from /Users/<user>/.platformio/packages/framework-arduinointel/system/libarc32_arduino101/framework/include/os/os.h:34:0,
from /Users/<user>/.platformio/packages/framework-arduinointel/system/libarc32_arduino101/framework/include/cfw/cfw.h:37,
from /Users/<user>/.platformio/packages/framework-arduinointel/system/libarc32_arduino101/framework/include/services/ble/ble_service_gap_api.h:34,
from /Users/<user>/.platformio/packages/framework-arduinointel/system/libarc32_arduino101/framework/include/services/ble/ble_service_gatt.h:35,
from /Users/<user>/.platformio/packages/framework-arduinointel/libraries/CurieBLE/src/BLECommon.h:26,
from /Users/<user>/.platformio/packages/framework-arduinointel/libraries/CurieBLE/src/BLEAttribute.h:23,
from /Users/<user>/.platformio/packages/framework-arduinointel/libraries/CurieBLE/src/BLECharacteristic.h:23,
from /Users/<user>/.platformio/packages/framework-arduinointel/libraries/CurieBLE/src/CurieBLE.h:20,
from src/Ada_CurieBot_RC.cpp:22:
/Users/<user>/.platformio/packages/framework-arduinointel/system/libarc32_arduino101/framework/include/os/os_types.h:42:0: note: this is the location of the previous definition
#define UNUSED(p) (void)(p)
^
In file included from .piolibdeps/BLEPeripheral_ID259/src/BLEPeripheral.h:10:0,
from src/Ada_CurieBot_RC.cpp:23:
.piolibdeps/BLEPeripheral_ID259/src/BLECentral.h:9:7: error: redefinition of ‘class BLECentral’
class BLECentral
^
```

and so on with similar errors ...

vojtamolda, 2017-06-18:
Hello @anweiss,

I think your problem isn't caused _PlatformIO_. You're mixing `BLEPeripheral` and `CurieBLE` libraries in your project. Both libraries have nearly identical API, but you can't use them together to build _Arduino 101_ code. Having `BLEPeripheral` installed causes the double definition problems because Arduino Intel build framework has the classes already included.

Just replace all the `#include` directives to use `CurieBLE` instead and delete `BLEPeripheral` from your _PlatfromIO_ build and you should be able to build without any issues.

anweiss, 2017-06-18:
Thanks @vojtamolda ... managed to build without issue ... however, for some reason, `BLEPeripheral::begin()` from CurieBLE does not seem to execute.

closed 2017-06-18 by anweiss