sandeepmistry/arduino-BLEPeripheral #164: Library function clearBondStoreData() overwrites bootloader on nRF52832

mholin1, 2017-06-22:
Running "HID_test" Example from the Arduino BLEPeripheral Library on a Sparkfun nRF52832 Breakout Board as-is but running HID_test with the function clearBondStoreData() in place apparently overwrites some part of the bootloader as the board will not then boot to the application after a reset resulting in the the Softdevice, application and bootloader having to be reloaded.

bojh, 2017-06-26:
I had the same issue working with a modified OTA bootloader on nRF52832 from the nRF5_SDK_11.0.0/examples/dfu/bootloader. You can shift your bondstore by 6 code pages, then the bondstore is starting from address: 0x79000
`  BLEBondStore    bondStore = BLEBondStore(6);  
`
... before binding by e.g.:
`BLE.setBondStore(bondStore);`

No arguments in the BondStore constructor does use the `offset=0` (default) resulting to a bondstore start address 0x7F000. `offset=6` does result in the start address 0x79000. The DFU bootloader area is located to 0x7A00..0x8000. 
See also the memory map:
[http://infocenter.nordicsemi.com](http://infocenter.nordicsemi.com) Software Development Kit > nRF5 SDK > nRF5 SDK v11.0.0 > Examples > DFU bootloader examples > BLE & HCI/UART Bootloader/DFU

mholin1, 2017-06-26:
Man you are a life saver.  Admittedly don't know enough about C++ as I should.  That fixed the overwrite problem as well as the problem with Win. 10 dropping the connection (see my other issue 160).

Our paths ever cross I buy you a drink, lol. 

--------------------------------------------

 

 
 
 
 
 
 http://infocenter.nordicsemi.com
 
 it on GitHub, or mute
 the thread.

bojh, 2017-06-26:
you are welcome  - that's the benefit of open source

mholin1, 2017-06-26:
A colleague once said engineering is the process of knocking down a brick wall only to find another just behind it.  It is working better after moving the bond store but something is still up as the board still does not boot to the application after hardware reset requiring the firmware to be reloaded.  Also unfortunately the Win. 10 Driver Error (issue 160) is still there.    Still much appreciate the progress.

sandeepmistry, 2017-07-31:
Closing as resolved, feel free open a pull request that changes the flash page number for the Sparkfun board.

closed 2017-07-31 by sandeepmistry

mholin1, 2017-08-02:
Unfortunately changing the flash page helps but there is still an overwrite problem as the board does not reset properly.  I can somewhat get around it by removing the bootloader and just flashing directly with a JLink board.

The real issue though is still the Windows 10 Driver Error.  (Issue 160)  If anyone could give any insight into that it would be most appreciated.

Thanks,
Mark O.

--------------------------------------------