sandeepmistry/arduino-BLEPeripheral #169: Temp example not working

gesaleh, 2017-07-22:
Hi 
I have a  blend 2 I'm trying to use the library but it is failing for issue with timer
```
WARNING: library TimerOne claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (nRF5) architecture(s).
temp_sensor:140: error: 'class TimerOne' has no member named 'initialize'
   Timer1.initialize(2 * 1000000); // in milliseconds
          ^
temp_sensor:141: error: 'class TimerOne' has no member named 'attachInterrupt'
   Timer1.attachInterrupt(timerHandler);
          ^
exit status 1
'class TimerOne' has no member named 'initialize'
```

is there a specific path for this lib ? to work with nrf5 ?

gesaleh, 2017-07-23:
I'm using arduino 1.8.3 and BLE Blend 2

sandeepmistry, 2017-07-31:
This example only works on AVR boards, as the TimerOne library only supports AVR boards.

closed 2017-07-31 by sandeepmistry