sandeepmistry/arduino-BLEPeripheral #170: BLESrerial stream libray issue

cheeta1, 2017-08-02:
BLESerial.parseFloat();
BLESerial.parseInt();
these function are not working properly they only return the value of firts two char that i send eg if i send 1000 it return 10. i have tried to increase timeout to 5 sec but still same results

mlu, 2017-08-14:
There is a problem in BLESerial::peek() that reads one byte off, should read the same byte as would be returned by a call to read(), just without advancing  _rxTail.

-  uint8_t byte = this->_rxBuffer[this->_rxTail];
+  uint8_t byte = this->_rxBuffer[ (this->_rxTail + 1) % sizeof(this->_rxBuffer)];

sandeepmistry, 2017-09-04:
Fixed via https://github.com/mlu/arduino-BLEPeripheral/commit/3e80f487662abe4474aa1cd663b8cf3554d694c8 from @mlu.

@mlu thanks for the patch, @cheeta1 could you please  try the Github master version of the lib?

closed 2017-09-04 by sandeepmistry

cheeta1, 2017-11-02:
hi there i have tested this but still having issues. now i only get first character i.e if i send 123 then i get 1. can you solve this issue?

cheeta1, 2017-11-02:
hi @sandeepmistry it is well thanks for the correction update.