sandeepmistry/arduino-BLEPeripheral #172: BLECharacteristic::setValue always echoes writes from Central back to Central

mlu, 2017-08-10:
When playing around with MIDI characteristic I found that every note sent from a Central (iMac MIDI) was echoed back, that is not always what is wanted. 

The reason is that the function setValue(const unsigned char value[], unsigned char length) that actually makes the change always invokes the listener callback. So when receiving a characteristic write from Central and we have writes without response the listener should not be invoked.

One possible way, that so far works for me is:

void BLECharacteristic::setValue(BLECentral& central, const unsigned char value[], unsigned char length) {
--  this->setValue(value, length);
++  if ( _properties & BLEWriteWithoutResponse ) {
++	this->_valueLength = min(length, this->_valueSize);
++	memcpy(this->_value, value, this->_valueLength);
++  }
++  else {
++	this->setValue(value, length);
++  }

  this->_written = true;

  BLECharacteristicEventHandler eventHandler = this->_eventHandlers[BLEWritten];
  if (eventHandler) {
    eventHandler(central, *this);
  }
}

sandeepmistry, 2017-08-12:
@mlu could you please provide a minimal sketch that demonstrates the issues, along with a step by step procedure to reproduce it. Thanks.

mlu, 2017-08-14:
Link to the sketch:

https://drive.google.com/open?id=0BwkzMBFZwrYtc1NKZ1ppTFRtUjQ

midiCharacteristic is configured BLEWriteWithoutResponse.
The sketch does not send any midi data, it only receives. So there should be no output.

1) Compiled and loaded to a Micro:Bit board.

2) Connect to a BLE MIDI enabled computer
    On a Mac computer, open Tools : Sound/MIDI settings  
    Double click Bluetooth Tool icon
    Connect to the MICRO:Midi ble unit

3) Use some midi generator to send MIDI notes to the board and monitor the traffic
    Use MIDI Keys , a minimal virtual MIDI keyboard and route output to the MICRO:Midi
    Use MIDI Monitor to record output from MICRO:Midi and also spy in traffic to the board.

Now every MIDI message sent to to board will be echoed back to the computer, this is not done by the application code in the sketch but by the BLE transport layer.

kargeor, 2020-12-29:
I have the same problem, please merge #174.

kargeor, 2020-12-29:
I have confirmed it works with BLE MIDI.