sandeepmistry/arduino-BLEPeripheral #178: problems with  led exaple and jlink what is going on .....

SoniaZotz, 2017-08-28:
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
Info : No device selected, using first device.
Info : J-Link V9 compiled Jul 24 2017 17:37:57
Info : Hardware version: 9.40
Info : VTarget = 3.325 V
Info : clock speed 1000 kHz
Info : SWD IDCODE 0x0bb11477
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x0001b4c0 msp: 0x20003fc0
Warn : Unknown device (HWID 0x0000008f)
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
auto erase enabled
Info : Padding image section 0 with 2112 bytes
Warn : not enough working area available(requested 32)
Warn : no working area available, falling back to slow memory writes
wrote 110592 bytes from file C:\Users\walter\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0/cores/nRF5/SDK/components/softdevice/s130/hex/s130_nrf51_2.0.1_softdevice.hex in 40.920341s (2.639 KiB/s)
** Programming Finished **
** Verify Started **
Warn : not enough working area available(requested 52)
Warn : not enough working area available(requested 52)
verified 108448 bytes in 2.028116s (52.219 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

what is going on .....

mlu, 2017-08-28:
Seems that OpenOCD does not know how much RAM is available to run the fast programming and verification code on the nRF51 target, so it used the slower routines that makes remote writes to the  flash programming registers. Both programming an verification seems to have completed without problems, so everything seems ok.

What settings were used, commands was given to generate this output, so it can be reproduced ?

sandeepmistry, 2017-09-05:
I think this issue is in the wrong repo, there is nothing related to JLink or openocd in this Arduino Library.

closed 2017-09-05 by sandeepmistry