sandeepmistry/arduino-BLEPeripheral #179: Add support for BLE Nano V2  i.e. nRF52832

drmpf, 2017-09-02:
The latest Redbear BLE Nano (V2) uses nRF52832.  
Can you add support for this chip please

drmpf, 2017-09-05:
It appears that this board is supported via 
Arduino Core for Nordic Semiconductor nRF5 based boards
https://github.com/sandeepmistry/arduino-nRF5#supported-boards
I will try this process once the V2 board arrives.
Question "Can the DAPLink program RedBear provides in it KIT be used to Flash the SoftDevice instead of the CMSIS-DAP"?
if so should i use the RBL_DAPLink or the DAPLink  programmer?

sandeepmistry, 2017-09-11:
See pull request #185.

sandeepmistry, 2017-09-11:
@drmpf 

> Question "Can the DAPLink program RedBear provides in it KIT be used to Flash the SoftDevice instead of the CMSIS-DAP"?

Yes, I think CMSIS-DAP and DAP link are the same.

drmpf, 2017-09-11:
I confirm pull request [#185](https://github.com/sandeepmistry/arduino-BLEPeripheral/pull/185). 
works with RedBear Nano V2 and BLEPeripheral V0.4.0
I needed to add
```
  #include <WStream.h>
  #define Stream WStream
```
to a number of my files
Currently having trouble finding a define that correctly includes these two lines

drmpf, 2017-09-11:
I get the following warning but code works as expected

```
C:\Users\matthew\Documents\Arduino\SampleRedbearBLENanoScreens\SampleRedbearBLENanoScreens.ino:48:98: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]

     BLEService               uartService          = BLEService("6E400001B5A3F393E0A9E50E24DCCA9E");
```

drmpf, 2017-09-11:
```
#if defined( NRF52 )
  #include <WStream.h>
  #define Stream WStream
#else
  #include <Stream.h>
#endif
```
seems to work
`#if defined( ARDUINO_RBL_nRF52832 )`   DOES NOT

closed 2017-09-11 by drmpf