sandeepmistry/arduino-BLEPeripheral #186: no change in set tx power

Btje1977, 2017-09-14:
bool setTxPower(int txPower); was used and there is no change in power.

sandeepmistry, 2017-10-02:
@Btje1977 please share a minimal sketch to reproduce, and what board/chip set you are using.

Jhorvioke, 2017-10-13:
Hi. I'm trying to develop a bluetooth controlled automatic irrigation control. However, I have the same problem as Btje1977 ... when I call setTxPower there is no change in power. In my case I tried 2 generic Nrf51822 chips. Specifically, these 2 models: https://goo.gl/JJGUQ5 and https://goo.gl/hPi23D. I do not know if I misunderstood the setTxPower call method or missing something in my basic code:
`
#include <SPI.h>
#include <BLEPeripheral.h>
// define pins for nRF51822
#define BLE_REQ   -1
#define BLE_RDY   -1
#define BLE_RST   -1

#define pin_valve 30
#define pin_ledconnected 8

static int state = 0;
int counter = 0;
int wateringPower = -12;

BLEPeripheral               wateringdevice       = BLEPeripheral(BLE_REQ, BLE_RDY, BLE_RST);
BLEService           irrigationService           = BLEService("19b10000e8f2537e4f6cd104768a1214");
BLEDescriptor      valvedescriptor               = BLEDescriptor("2901", "valve state");
BLEIntCharacteristic    switchstatevalve         = BLEIntCharacteristic("19b10001e8f2537e4f6cd104768a1214", BLERead | BLEWrite);
BLERemoteService                 remoteGenericAttributeService            = BLERemoteService("1800");
BLERemoteCharacteristic          remoteDeviceNameCharacteristic           = BLERemoteCharacteristic("2a00", BLERead);

void setup() {
  Serial.begin(9600);
#if defined (__AVR_ATmega32U4__)
  delay(5000);
#endif

  pinMode(pin_valve, OUTPUT);
  pinMode(pin_ledconnected, OUTPUT);

  wateringdevice.setLocalName("wateringdevice 0001");
  wateringdevice.setAdvertisedServiceUuid(irrigationService.uuid());
  wateringdevice.setDeviceName("DevWaterx041");
  wateringdevice.setAppearance(0x0080);
  wateringdevice.addRemoteAttribute(remoteGenericAttributeService);
  wateringdevice.addRemoteAttribute(remoteDeviceNameCharacteristic);
  wateringdevice.addAttribute(irrigationService);
  wateringdevice.addAttribute(switchstatevalve);
  wateringdevice.addAttribute(valvedescriptor);
  wateringdevice.begin();
  wateringdevice.setTxPower(wateringPower);
}

void loop() {

  BLECentral central = wateringdevice.central();
  if (central) {
    while (central.connected()) {
      digitalWrite(pin_ledconnected, HIGH);
      // central still connected to peripheral
      if (switchstatevalve.written()) {
        state = switchstatevalve.value();
      }
    }
  }
  watering_control ();
}

void watering_control() {
  digitalWrite(pin_ledconnected, LOW);
  switch (state) {
    case 0:
      digitalWrite(pin_valve, LOW);
      break;
    case 1:
      digitalWrite(pin_valve, HIGH);
      break;
    case 2:
      if (counter < 3) {
        digitalWrite(pin_valve, HIGH);
        delay (10000);
        digitalWrite(pin_valve, LOW);
        delay (20000);
      } else {
        digitalWrite(pin_valve, LOW);
        counter = 0;
        state = 0;
      }
      counter++;
      break;
  }
}
`
Where am I going wrong?
Any help at all would be appreciated!

Btje1977, 2017-11-06:
Sorry for the delay:

// Import libraries (BLEPeripheral depends on SPI)
#include <SPI.h>
#include <BLEPeripheral.h>

// define pins (varies per shield/board)
#define BLE_REQ   10
#define BLE_RDY   2
#define BLE_RST   9

boolean ledState=LOW;

// create peripheral instance, see pinouts above
BLEPeripheral           blePeripheral        = BLEPeripheral(BLE_REQ, BLE_RDY, BLE_RST);
// create service
BLEService              ledService           = BLEService("19b10000e8f2537e4f6cd104768a1214");
// create switch characteristic
BLECharCharacteristic   switchCharacteristic = BLECharCharacteristic("19b10001e8f2537e4f6cd104768a1214", BLERead | BLEWrite);

void setup() {
  Serial.begin(9600);

  // set LED pin to output mode
 pinMode(PIN_LED1,OUTPUT);
 pinMode(PIN_LED2,OUTPUT);

  // set advertised local name
  blePeripheral.setLocalName("F1ST");
  
  // blePeripheral.addAttribute(ledService);
  blePeripheral.addAttribute(switchCharacteristic);
  
  // begin initialization
  blePeripheral.begin();

  //Different settings
  blePeripheral.setTxPower(-10);
  blePeripheral.setDeviceName("ServoTimer");
  
}

  
void loop() {

  //Bluetooth handeling
 BLECentral central = blePeripheral.central();

  //Alive led
  digitalWrite(PIN_LED1, HIGH);   
  delay(100);                      
  digitalWrite(PIN_LED1, LOW);   
  delay(10000); 
}

sandeepmistry, 2017-11-21:
@Btje1977 what board are you using?

sandeepmistry, 2018-01-08:
Closing for now due to lack of activity.

closed 2018-01-08 by sandeepmistry

rwadhwa25, 2018-02-14:
Hi,
I am facing the same issue as above. The board that I am using is Generic nRF51822. I'm setting the Tx Power as -30 but still do not see any change being reflected. Any reason for that/

Thank you!

TamojitSaha, 2018-09-26:
> 
> 
> Hi,
> I am facing the same issue as above. The board that I am using is Generic nRF51822. I'm setting the Tx Power as -30 but still do not see any change being reflected. Any reason for that/
> 
> Thank you!

I am facing the same issue.

TamojitSaha, 2018-09-27:
Now it works! Solved! Check (this)[#240] issue