sandeepmistry/arduino-BLEPeripheral #191: nRF51822 e-Paper Module (SPI)

Chop91, 2017-09-22:
Today I started a little project, a BLE e-paper display but I got stuck somewhere..

I bought the following e-paper:
http://www.waveshare.com/wiki/2.9inch_e-Paper_Module#Working_with_Arduino

I tried the demo code with my Arduino Uno and everything worked just fine, but if I try to use the e-paper with the nRF51822 nothing happens. I tried the original arduino pin configuration on the nRF51822 and also the nRF51822, since I read somewhere ( I think the reference manual) that these pins should be the correct ones.

e-Paper | Arduino | nRF51822
-- | -- | --
3.3V | 3V3 | 3V3
GND | GND | 3V3
DIN | D11 | 24
CLK | D13 | 25
CS | D10 | 30
DC | D9 | 9
RST | D8 | 8
BUSY | D7 | 7

But for both configurations the display does nothing.
Maybe someone knows what I am doing wrong or can provide a working example for a nRF51822 using an e-paper display?

dmikhalsky, 2017-09-25:
you have to look at varients.h pin definitions for Generic boards. You need SPI pins + a couple more

Chop91, 2017-09-25:
I bought the Waveshare BLE400 Kit from ebay, so I think I do not need the definition for generic boards. In the varients.h there is exactly the same pin definition as in my first posting.

e-Paper | SPI | Arduino Uno | nRF51822
-- | -- | -- | --
DIN | MOSI | D11 | 24
CLK | SCK | D13 | 25
CS | SS | D10 | 30

The remaining pins 7-9 should be fine too.

dmikhalsky, 2017-09-25:
I briefly looked at Waveshare wiki, it looks like they have their own protocol (not SPI)  that might be simply incompatible with nrf5x boards

sandeepmistry, 2017-10-02:
@Chop91 please only open issues related to this library in this issue tracker.

closed 2017-10-02 by sandeepmistry