sandeepmistry/arduino-BLEPeripheral #194: DFU OTA

mtiutiu, 2017-10-05:
Hi,

I hope that I'm using the right place for this kind of question: is it possible to have OTA support over BLE? I'm a newbie regarding BLE in general so please don't be too harsh with me:). I just started learning it and I found this library to be so great so thank you for the big effort on providing it(it's the most user friendly library that I found so far).

Thanks.

bojh, 2017-10-05:
Hello @mtiutiu 
The way you ca achieve that functionality is to setup your own customized bootloader including the DFU OTA service with a specific DFU button matching to your hardware situation. You can do that by cloning & modifing the bootloader example inside the nRF5_SDK_11.0.0/examples/dfu/bootloader. Ensure that you have a functional arm-gcc build chain using e.g. 2015q4. (make, gcc). You have to adapt  your <nrf5_SDK_11.0.0>\components\toolchain\gcc\Makefile.windows configuration (e.g Windows) to your installation pathes. To build, go to pca10040/dual_bank_ble_s132/armgcc and try to build (make)! With the mergehex.exe tool from nordic you can combine your s132_nrf52_2.0.0_softdevice.hex with your new bootloader. Using a Nordic nRF52-DK board (or Segger J-Link) you can put that via the (extern) debugger port to your target platform via drag&drop. To setup a DFU package zip-file from your compiled application hex-file you need the nrfutil.exe (version 052). Now your are ready at all: With the Nordic "nRF Tool" App "DFU" section you can load your new application firmware zip package. Puting your device in bootloader mode (supporting the DFU-OTA service) shows you a "DfuTarg" service device in your App. You can than start upload the new firmware package.

mtiutiu, 2017-10-05:
@bojh Thank you for the very detailed explanation, I appreciate. Will follow your directions and learn more on the way. Thanks a lot.

closed 2017-10-05 by mtiutiu