sandeepmistry/arduino-BLEPeripheral #199: not advertising ble devicenRf51822 waveshare(BLE400)

manishak123, 2017-10-30:
Hi,
When I am using separately BLEpepheral for advertising it is working fine with a mobile app. And if I am using MySensor.h separately MySensor it is also working fine with the gateway device.

In the client side am using the nRF51822 device waveshare (BLE400) and I am giving condition like if button state is LOW the client device should connect gateway device (nrf24L01 and ESP8266 nodeMCU). if the button is HIGH it should disconnect from the gateway device and it should start advertising and it should connect to the mobile app (NRF connect app), through the mobile app I am sending command 0 and 1 to switch ON and OFF the led of a client device.
At the client side, I am using both libraries like BLEpepheral.h for advertising and connect to the mobile app and MySensor.h library for connecting to the gateway and communicate with the gateway device. When I am executing this program client is communicating with gateway when pressing the button is HIGH  the gateway is not communicating with the client , but the client is not advertising to connect mobile app( (NRF connect app). Why is a ble device not advertising?

sandeepmistry, 2017-11-13:
@manishak123 do you have a link to the MySensor library? Does it also use Bluetooth functionality?

manishak123, 2017-11-13:
Yes sir, "https://github.com/mysensors/MySensors" in this library Bluetooth functionality like regarding advertising is not here that's why I used your BLEPepheral.h library and MySensor.h library for advertising purpose. I created this program for switching functionality like client connecting to gateway device when the switch is low and client should advertise when the switch is high. but after the switch is the high client is not able to advertise. I don't know what is the exact reason behind.

[ble_mysensor.zip](https://github.com/sandeepmistry/arduino-BLEPeripheral/files/1466099/ble_mysensor.zip)

sandeepmistry, 2017-11-21:
So if the MySensors library is also accessing Bluetooth functionality, it will conflict with this one. I cannot recommend using both at the same time.

closed 2017-11-21 by sandeepmistry