sandeepmistry/arduino-BLEPeripheral #205: Conflict with attachInterrupt() on NRF5 Core (NRF52832)?

alextsam, 2017-12-08:
Hi,

I was not sure whether to post this here or to the NRF5 Arduino Core repo, but I decided to do it here as this is the part that gets impacted.
The issue is that the stack doesn't seem to work when there is a gpio interrupt attached before the beacon begin(). When I say "doesn't seem to work" I mean that there are no advertisements (tested with eddystone URL). To make things weirder, everything is fine if the attachInterrupt() is moved after the beacon begin(). 
Example sketch:

```
#include <EddystoneBeacon.h>

#define BUTTON (8u)

EddystoneBeacon eddystoneBeacon = EddystoneBeacon();
volatile bool btnPressed = false;

void btnPress(void)
{
    if (!digitalRead(BUTTON)) {
        btnPressed = true;
    }
}

void setup()
{
  pinMode(BUTTON, INPUT_PULLUP);
  attachInterrupt(BUTTON, btnPress, CHANGE);
  
  eddystoneBeacon.begin(-18, "http://www.example.com"); // power, URI
}

void loop()
{
  eddystoneBeacon.loop();
}
```

So, with this sketch, nothing seems to come out of the beacon. If you just move the attachInterrupt() line 
 to be under eddystoneBeacon.begin() then everything is fine. Different kinds of interrupt (eg FALLING) didn't make any difference.

While trying to find what the issue is, I found that just enabling the interrupts (NVIC_EnableIRQ(GPIOTE_IRQn)) before eddystoneBeacon.begin() is what "breaks" it.

Even though, as I said, I have a workaround for my application, I would love to get this fixed properly for everyone. Any ideas?
Thanks.

Alex

sandeepmistry, 2018-01-08:
Does any code after the `attachInterupt` execute?

alextsam, 2018-01-08:
Yes, this issue came up in a larger application that had a bunch of other things happening after the attachInterrupt() and everything else is fine (the code in the ISR as well). I made the sketch posted above to help with replicating the issue. You can just copy-paste it and swap the lines to make it work as I mention in the original post.

mristau, 2018-02-22:
Thanks a lot, this helped me a lot, i had the same issue and couldn't find any error in my code.
When i made an empty test sketch, all my code worked, but in my main sketch BLE never showed up, everything else works great.
After moving my BLE-Init before the first interrupt, everything works.

ldab, 2019-03-27:
+1 it does not like `blePeriph.begin();` at all.

sandeepmistry, 2020-11-27:
Fixed via @nobodyguy's PR https://github.com/sandeepmistry/arduino-nRF5/pull/422.

closed 2020-11-27 by sandeepmistry