sandeepmistry/arduino-BLEPeripheral #206: BLE HID keyboard, multimedia reconnection issue

cheeta1, 2017-12-28:
i have been working on BLE HID keyboard and multimedia. i have used  one of your keyboard example. first time the keyboard works perfectly. its pairs with my iphone and works nicely but when i power off and then power on the nrf51822 it connects to my iphone automatically but do not send the key data to iphone the debug is given below

//////////////////////first startup after programming////////////////////////////////

version = 8 89 135
Start advertisement
Peripheral address: 
BLE HID Volume Knob
Evt Connected 69:5b:bd:9e:90:3c
Peripheral connected to central: 69:5b:bd:9e:90:3c
Connected to central: 69:5b:bd:9e:90:3c
Evt Write, handle = 11
02 00 
Evt Sec Params Request 1 1 4 0 0 16
Evt Conn Sec Update 1 2 16
Evt Auth Status
0
Storing bond data
Peripheral bonded: 69:5b:bd:9e:90:3c
Evt Write, handle = 21
01 00 
Evt Conn Param Update 0xC 0xC 0x4 0x64
Evt TX complete 4
Evt TX complete 2
Evt TX complete 2
Evt TX complete 4
Evt TX complete 2
Evt TX complete 2

/////////////////////////power off and on the nrf51822/////////////////////

version = 8 89 135
Start advertisement
Peripheral address: 
BLE HID Volume Knob
Evt Connected 69:5b:bd:9e:90:3c
Peripheral connected to central: 69:5b:bd:9e:90:3c
Connected to central: 69:5b:bd:9e:90:3c
Evt Sec Info Request 27753 1 0 0
Evt Write, handle = 11
02 00 
Evt Sec Info Request 27753 1 0 0

and the serial monitor does not show anything else.

[image]

any idea why it is hapenning

RicoKirstein, 2017-12-29:
Is your example calling bleHIDPeripheral.clearBondStoreData(); ? had this happen before.

cheeta1, 2018-01-01:
hi @RicoKirstein  yes i am calling clearBondStore and it always happen.

RicoKirstein, 2018-01-02:
@cheeta1 if you clear the bond store on every startup, it can't reconnect to your phone. The phone still remembers but the nrf won't. You need to not clear it.

cheeta1, 2018-01-02:
hi @RicoKirstein the nrf connects to the phone automatically as i have provided in the debug above but do not send the key press data. i have tried removing the clear bond store but the issue remains same.

RicoKirstein, 2018-01-02:
for how long are you disabling power to the nrf? and which dev board are you using? I only experimented with the keyboard example. It worked on android, but if you restart it too fast, maybe the phone doesn't notice that it has to reregister the notification events. The interesting thing is on android the phone wouldn't reconnect at all when not deleting bonding storage. Are you using a debugger or the bootloader? There seems to be an issue with the combination of the bootloader and the bond storage.
 I moved from arduino to the official nordic sdk, and except for the red bear boards not being compatible out of the box, it seems to work much better.

cheeta1, 2018-01-02:
few seconds sometimes just plug in and out

cheeta1, 2018-01-06:
hi there i have managed to work it on iphone flawlessly but still having issues on  Android. it seem that android deos not send the data requested by the nrf51822. when i turn off and on the bluetooth on Android  it gets connected immediately but no communication done between nrf51822 and Android. 

can anyone help me out with this issue?

cheeta1, 2018-01-07:
hi there here is my code 

// Copyright (c) Sandeep Mistry. All rights reserved.
// Licensed under the MIT license. See LICENSE file in the project root for full license information.

// Import libraries (BLEPeripheral depends on SPI)
#include <SPI.h>
#include <BLEHIDPeripheral.h>
#include <BLEKeyboard.h>

#define ANDROID_CENTRAL

//custom boards may override default pin definitions with BLEHIDPeripheral(PIN_REQ, PIN_RDY, PIN_RST)
BLEHIDPeripheral bleHIDPeripheral = BLEHIDPeripheral();
BLEKeyboard bleKeyboard;

void setup() {
  Serial.begin(9600);
  pinMode(23,INPUT_PULLUP);
  pinMode(24,INPUT_PULLUP);
  pinMode(25,INPUT_PULLUP);
  pinMode(28,INPUT_PULLUP);
  
#if defined (__AVR_ATmega32U4__)
  while(!Serial);
#endif

  // clear bond store data
  bleHIDPeripheral.clearBondStoreData();

#ifdef ANDROID_CENTRAL
  bleHIDPeripheral.setReportIdOffset(1);
#endif

  bleHIDPeripheral.setLocalName("HID Keyboard");
  bleHIDPeripheral.addHID(bleKeyboard);

  bleHIDPeripheral.begin();

  Serial.println(F("BLE HID Keyboard"));
}

void loop() {
  BLECentral central = bleHIDPeripheral.central();

  if (central) {
    // central connected to peripheral
    Serial.print(F("Connected to central: "));
    Serial.println(central.address());

    while (central.connected()) {

      if(digitalRead(23)==LOW)
      {

        bleKeyboard.press(KEYCODE_A);
        delay(50);
        bleKeyboard.release(KEYCODE_A);
        delay(500);
      }
      if(digitalRead(24)==LOW)
      {

       bleKeyboard.press(KEYCODE_B);
        delay(50);
        bleKeyboard.release(KEYCODE_B);
        delay(500);
      }
      if(digitalRead(25)==LOW)
      {

        bleKeyboard.press(KEYCODE_C);
        delay(50);
        bleKeyboard.release(KEYCODE_C);
        delay(500);
      }
      if(digitalRead(28)==LOW)
      {

        bleKeyboard.press(KEYCODE_D);
        delay(50);
        bleKeyboard.release(KEYCODE_D);
        delay(500);
      }
     
    }

    // central disconnected
    Serial.print(F("Disconnected from central: "));
    Serial.println(central.address());
  }
}

and here is the debug data

Evt Connected 7a:15:d2:a7:83:78
Connected to central: 7a:15:d2:a7:83:78
Evt Sec Params Request 1 1 4 0 0 16
Evt Conn Param Update 0x6 0x6 0x0 0x7D0
Evt Conn Sec Update 1 2 16
Evt Auth Status
0
Storing bond data
Evt Conn Param Update 0x27 0x27 0x0 0x7D0
Evt Conn Param Update 0x6 0x6 0x0 0x7D0
Evt Write, handle = 11
02 00 
Evt Conn Param Update 0x27 0x27 0x0 0x7D0
Evt Write, handle = 21
01 00 
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1
Evt TX complete 1

###android bluetooth turned of and the on ########

Evt Disconnected
Start advertisement
Disconnected from central: 7a:15:d2:a7:83:78
Evt Connected 62:26:44:e1:85:47
Connected to central: 62:26:44:e1:85:47
Evt Conn Param Update 0x6 0x6 0x0 0x7D0
Evt Sec Info Request 28592 1 0 0
Evt Conn Sec Update 1 2 16
Evt Conn Param Update 0x27 0x27 0x0 0x7D0

after that if i press any key no debug data appears and andriod dos not respond although it is connected to nrf51822 device

need a little help with this thanks!

cheeta1, 2018-01-18:
ok i have tried with different mobiles and here are the results

nexus 5 ---> andriod 6.0.1  ----> after reconnection not working
samsung S4 ---> andriod 5.0.1  ----> after reconnection not working
nexus 6P ---> andriod 8.1.0  ----> after reconnection  working sucessfully
samsung s8 ---> andriod 7.0  ----> after reconnection working sucessfully

elRadish, 2019-07-05:
I have exactly the same problem with a cjmcu-8223 with QFAA-H0 chip. I'm connecting to a MacBook. First connection after programming is fine. Power cycle, than reconnection, but no more data.
Any advice?

jczhangwbb, 2023-10-14:
I am in the same situation as you. I am using nRF51822 and have combined both mouse and keyboard functions in HID, but I am using it on a PC. Everything worked fine the first time, but when I power cycle the device, the mouse functions work correctly, but when I try to send keyboard input, I get an error code 8. According to the definition in nrf_error.h, this is NRF_ERROR_INVALID_STATE, which means "Invalid state, operation disallowed in this state." This is puzzling because the implementation process for the keyboard is identical to that of the mouse, so if it were a state issue, the mouse should also have the same problem, but it doesn't. Similarly, if I delete the pairing information and re-pair, it works again. I'm completely stumped. I don't know if you've made any new progress on your end. Thx!