sandeepmistry/arduino-BLEPeripheral #208: Starter Sketch not working on NRF52-DK board

hongkongkiwi, 2018-01-18:
I'm having trouble getting this library working with the NRF52832 development board (ntf52-dk).

I've tried a simple blinky example by myself and it works fine, however when I try to use any Bluetooth functions in this library (e.g. using the example sketch), it doesn't advertise.

It compiles and uploads fine, I have also uploaded the 132 soft device.

sandeepmistry, 2018-01-19:
@hongkongkiwi is it just the starter sketch broken?

Have you tried any other examples like the LED one? https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/led/led.ino

hongkongkiwi, 2018-01-19:
So, just did some more testing. Interestingly when I select Nordic Semiconductor NRF52 DK as the board type it doesn't work. I can't even toggle an LED in setup.

However, when I select Generic NRF52, I can now toggle an LED during setup. However running the above code I cannot get it advertising at all, although I can still toggle an LED during setup.

I am using the latest version of the NRF52 DK board (2017).

utiq, 2019-08-08:
Could you solve this problem?, I have the same

viscount-monty, 2019-08-19:
I seem to have the same issue, any progress updates?

utiq, 2019-08-19:
I solved it flashing the module with the softdevice version 2.0.1

viscount-monty, 2019-08-20:
Thanks for the response - interesting, I remember that flashing that particular version of the softdevice allowed the sketches I uploaded to run and manipulate the GPIO. Are you saying that uploading that version of the softdevice made the BLE sketches work? Which sketch works for you?

My module is currently flashed with s132_nrf52_2.0.1_softdevice.hex downloaded from Nordic Semiconductor.

My issue seems identical to the original post - I can upload a simple sketch that correctly manipulates the GPIO pins, but none of the example BLE sketches seem to make the device advertise to be picked up on a scan (nRF Connect app - which picks up other BLE devices)

Seems so close, but so far!

viscount-monty, 2019-08-20:
SUCCESS UPDATE: 

For my module [Raytac MDBT42Q] I needed to select the 'RC Oscillator' option for Low Frequency Clock.

I can now see the device on BLE scan, connect to it. and change characteristics.