sandeepmistry/arduino-BLEPeripheral #215: Example sketch to send battery voltage every x seconds with nRF5x

probonopd, 2018-02-17:
Please add an example sketch to send battery voltage every x seconds with nRF5x.

The sketch should be optimized for low power consumption to run off a coin cell for a long time.

I think this kind of example will be very useful for anyone who wants to make a sensor that sends data in intervals from battery-operated sensor devices.

probonopd, 2018-02-17:
Maybe @d00616 could help, given your [apparent knowledge](https://github.com/sandeepmistry/arduino-nRF5/issues/153#issuecomment-299954663) about sleep modes on the nRF5x?

probonopd, 2018-02-17:
Also relevant: https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/203. Unfortunately this was closed because it would need "changes to library".

So this ticket is here to request those changes.

mristau, 2018-02-18:
i found this sketch linked somewhere in the issues here
https://github.com/kriswiner/nRF52832DevBoard/blob/master/BMP280_nRF52.ino
there is a BLE BatteryService implemented and it will update the battery reading in loop.

He also has a description of the voltage measurement at the comments on top

probonopd, 2018-02-18:
Thanks @mristau for linking the example sketch of @kriswiner. He writes

> A resistor divider needs to be added to any analog pin so the voltage can be sampled 
 *  between 0 and the 3V3 of the board (27K/100K will do) such that the ADC will read 0 at 0 V and 4095 at 
 *  4.2 V of LiPo battery voltage.

Since I assume this is of interest to many users, I'd suggest to include a stripped-down version of his sketch (only the battery part) as one of the official examples.

probonopd, 2018-02-18:
So what I am essentially looking for is an Arduino version of
https://github.com/NordicPlayground/nrf51-ble-app-temp

> This project takes a temperature measurement using the internal temperature sensor of the nRF51822, does an ADC reading to get the battery voltage, packs it up as service data in an advertisement packet and sends it, along with an ID string based on the Bluetooth address of the chip. This is the format expected by the nRF Temp apps for iOS and Android, so these will be capable of picking this up and showing the measurement.

probonopd, 2018-02-18:
I also see the functions 

https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/161a4163f565be3cd5b62bbc59f0c2b522d82b02/src/nRF8001.h#L81-L82

We should make an example sketch using them for nRF5.

probonopd, 2018-02-18:
This is what I have at the moment - questions inline:

```
/* Partly from https://github.com/zebular13/BLE_Thermometer/blob/master/BLE_Thermometer_final/BLE_Thermometer_final.ino
   and from http://mochikun.hatenablog.com/entry/2016/03/28/genuino-101-dht22-%25e3%2581%25a7%25e6%25b8%25a9%25e5%25ba%25a6%25e6%2583%2585%25e5%25a0%25b1%25e3%2582%2592ble%25e3%2581%25a7%25e9%2580%2581%25e3%2582%258b%25ef%25bc%2588%25e3%2581%25aa%25e3%2582%2593
   and from https://github.com/sandeepmistry/arduino-BLEPeripheral/pull/115
   Shows 32 degrees for me in the nRF Connect app, and nRF Toolbox HTM does connect to it but not show the temperature - why?
*/

#include <BLEPeripheral.h>

BLEPeripheral blePeripheral;
BLEService tempService("1809");  // Health Thermometer Service
BLECharacteristic tempChar("2A1C", BLERead | BLENotify, 5); // Temperature Measurement

long previousMillis = 0;  // last time the temperature was checked, in ms

void setup() {

  blePeripheral.setLocalName("TempSketch");
  blePeripheral.setAdvertisedServiceUuid(tempService.uuid());
  blePeripheral.addAttribute(tempService);
  blePeripheral.addAttribute(tempChar);
  blePeripheral.begin();
}

void loop() {
  BLECentral central = blePeripheral.central();
  if (central.connected()) {
    // If 
    long currentMillis = millis();
    // If a central is connected to this peripheral and some time has passed, 
    // measure and update the temperature. 
    // Does it make a difference if we do this only if there is a significant change?
    if (currentMillis - previousMillis >= 5000) {
      previousMillis = currentMillis;
      int32_t rawTemperature = 0;
      sd_temp_get(&rawTemperature); // Does this work? I get 32 degrees
      float t = rawTemperature / 4.0;
      int it = (int)t;
      // How can we determine the following line from the BLE spec?
      const unsigned char vals[5] = {0, (char)t, 0, 0, 0}; // Incorrect, we are losing the decimals here?
      tempChar.setValue(vals, 5);
    }
  }
}
```

mristau, 2018-02-19:
I didn't work with that temperature sensor, we have one embedded into our accelerometer.

There has to be some connection from battery to a nRF pin, at our board it's to A2, then analogRead(PIN) will get a value between 0 and 1023, or 4095 if using 12bit
0 is 0V or LOW, 4095 is 3.3V or HIGH, with your resistor divider you can calculate the voltage of the battery itself like @kriswiner did it

The voltage divider should be chosen by the Voltage of the battery, at LiPo 3.7-4.2V are max values. Those should be reduced to below 3.3V
Our board designer chose a 47K / 120K divider for this.

closed 2018-02-22 by sandeepmistry

probonopd, 2018-02-23:
While this thread is very useful (thanks @mristau) I still would like to see an example sketch to be added so that one can get this kind of thing right from the examples rather than having to search for it on GitHub. Hence I would appreciate if you could give it consideration @sandeepmistry. I think making _low power devices_ that can also send their own battery level and temperature is one of _the_ killer applications for this project.

Autobot86, 2018-08-29:
How to add BLE batteryService code add to Beacon concept

TamojitSaha, 2018-09-24:
@probonopd 
> This is what I have at the moment - questions inline:
> 
> ```
> /* Partly from https://github.com/zebular13/BLE_Thermometer/blob/master/BLE_Thermometer_final/BLE_Thermometer_final.ino
>    and from http://mochikun.hatenablog.com/entry/2016/03/28/genuino-101-dht22-%25e3%2581%25a7%25e6%25b8%25a9%25e5%25ba%25a6%25e6%2583%2585%25e5%25a0%25b1%25e3%2582%2592ble%25e3%2581%25a7%25e9%2580%2581%25e3%2582%258b%25ef%25bc%2588%25e3%2581%25aa%25e3%2582%2593
>    and from https://github.com/sandeepmistry/arduino-BLEPeripheral/pull/115
>    and from https://github.com/kriswiner/nRF52832DevBoard/blob/master/BMP280_nRF52.ino
>    Shows 32 degrees for me in the nRF Connect app, and nRF Toolbox app HTM does connect to it but not show the temperature - why?
> */
> 
> #include <BLEPeripheral.h>
> 
> // Health Thermometer Service
> BLEPeripheral blePeripheral;
> BLEService tempService("1809");
> BLECharacteristic tempChar("2A1C", BLERead | BLENotify, 5); // Temperature Measurement
> 
> // Environmental Sensing Service
> BLEService env_sensingService = BLEService("181A");
> BLEShortCharacteristic tempCharacteristic = BLEShortCharacteristic("2A6E", BLERead | BLENotify); // temperature is int16_t
> 
> // Battery Service
> BLEService batteryService = BLEService("180F");
> BLEUnsignedCharCharacteristic battlevelCharacteristic = BLEUnsignedCharCharacteristic("2A19", BLERead | BLENotify); // battery level is uint8_t
> 
> long previousMillis = 0;  // last time the temperature was checked, in ms
> 
> void setup() {
> 
>   blePeripheral.setLocalName("TempSketch");
>   
>   // Health Thermometer Service
>   blePeripheral.setAdvertisedServiceUuid(tempService.uuid()); // Note: must only be used once, for the "main" service
>   blePeripheral.addAttribute(tempService);
>   blePeripheral.addAttribute(tempChar);
> 
>   // Battery service
>   blePeripheral.addAttribute(batteryService);
>   blePeripheral.addAttribute(battlevelCharacteristic);
>   
>   // Environmental Sensor service
>   blePeripheral.addAttribute(env_sensingService);
>   blePeripheral.addAttribute(tempCharacteristic);
> 
>   blePeripheral.begin();
> }
> 
> void loop() {
>   BLECentral central = blePeripheral.central();
>   if (central.connected()) {
>     // If 
>     long currentMillis = millis();
>     // If a central is connected to this peripheral and some time has passed, 
>     // measure and update the temperature. 
>     // Does it make a difference if we do this only if there is a significant change?
>     if (currentMillis - previousMillis >= 5000) {
>       previousMillis = currentMillis;
>       int32_t rawTemperature = 0;
>       sd_temp_get(&rawTemperature); // Does this work? I get 32 degrees
>       float t = rawTemperature / 4.0;
>       int it = (int)t;
>       // How can we determine the following line from the BLE spec?
>       const unsigned char vals[5] = {0, (char)t, 0, 0, 0}; // Incorrect, we are losing the decimals here?
>       tempChar.setValue(vals, 5);
>       uint8_t battlevel = 50; // Battery level from 0 to 100 - how do I get the real value?
>       battlevelCharacteristic.setValue((uint8_t)battlevel); // Set battery level value
>       int16_t reading = it * 100;
>       tempCharacteristic.setValue((int16_t)(reading)); // Needs temperature as degrees celsius x 100
>     }
>   }
> }
> ```

Any idea about to send string data from app(nrfToolBox) to ble module([nrf51822](https://potentiallabs.com/cart/image/cache/catalog/New%20Components/NRF51822%20BLE-800x800.jpg)) and vice-versa?

Praneeth2498, 2019-02-19:
can i get the code for nrf52832 Development kit