sandeepmistry/arduino-BLEPeripheral #217: Include BLEPeripheral nRF51822 - Low Power (Interrupt) Example

probonopd, 2018-02-18:
https://gist.github.com/sandeepmistry/6a6dfe9021b19bfddd60 should be included in the `examples/` that come with the library. Many people will be interested in a Low Power (Interrupt) Example.

sandeepmistry, 2018-02-22:
This only works with nRF5x boards, so not too keen.

closed 2018-02-22 by sandeepmistry

probonopd, 2018-02-23:
Is it a requirement that all example sketches need to run on boards other than nRF5x as well? That would limit the power of this project in my opinion.

Can't we have it print a statement if used with an incompatible board?

I assume that many people are using this project exactly for the reason that they want to make something _low power_, and a bare nRF5x board is _best_ for this. In fact, that's why I use this project, anyway ;-)

kainhofer, 2018-10-08:
I would also like to ask for including the example with the library. To indicate that it works only with the nrf5x boards, simply include nrf51 into the name of the example and it should be crystal clear that the example is only for specific boards...