sandeepmistry/arduino-BLEPeripheral #222: temp_sensor issues with Library "timerone"

ajaybnl, 2018-03-26:
**Timerone** library is not ready for Nrf51 Boards.

Please remove this library from your source code and add millis timer function.

Errors:

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:43:4: error: expected constructor, destructor, or type conversion before '(' token

 ISR(TIMER1_OVF_vect)          // interrupt service routine that wraps a user defined function supplied by attachInterrupt

    ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp: In member function 'void TimerOne::initialize(long int)':

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:51:3: error: 'TCCR1A' was not declared in this scope

   TCCR1A = 0;                 // clear control register A 

   ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:52:3: error: 'TCCR1B' was not declared in this scope

   TCCR1B = _BV(WGM13);        // set mode 8: phase and frequency correct pwm, stop the timer

   ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:52:16: error: 'WGM13' was not declared in this scope

   TCCR1B = _BV(WGM13);        // set mode 8: phase and frequency correct pwm, stop the timer

                ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:52:21: error: '_BV' was not declared in this scope

   TCCR1B = _BV(WGM13);        // set mode 8: phase and frequency correct pwm, stop the timer

                     ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp: In member function 'void TimerOne::setPeriod(long int)':

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:61:62: error: 'CS10' was not declared in this scope

   if(cycles < RESOLUTION)              clockSelectBits = _BV(CS10);              // no prescale, full xtal

                                                              ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:61:66: error: '_BV' was not declared in this scope

   if(cycles < RESOLUTION)              clockSelectBits = _BV(CS10);              // no prescale, full xtal

                                                                  ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:62:62: error: 'CS11' was not declared in this scope

   else if((cycles >>= 3) < RESOLUTION) clockSelectBits = _BV(CS11);              // prescale by /8

                                                              ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:62:66: error: '_BV' was not declared in this scope

   else if((cycles >>= 3) < RESOLUTION) clockSelectBits = _BV(CS11);              // prescale by /8

                                                                  ^

C:\Users\sid\Documents\Arduino\libraries\Timerone\TimerOne.cpp:63:62: error: 'CS11' was not declared in this scope

   else if((cycles >>= 3) < RESOLUTION) clockSelectBits = _BV(CS11) | _BV(CS10);  // p

labajo, 2018-04-09:
I tried other solutions as SimpleTimer but I couldn't have it working. I'd like to use a Timer with a callback. I don't think it is a good practice to count in the loop function.

labajo, 2018-04-10:
I think we could use the solution exposed in #215

closed 2022-11-20 by ajaybnl