sandeepmistry/arduino-BLEPeripheral #223: Joystick implementation and Example

marcotchella, 2018-03-29:
I deploy to my board  the example [HID_joystick_mouse.ino](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/HID/HID_joystick_mouse/HID_joystick_mouse.ino). In Android Smartphone the device is recognized as a Mouse. Looking the descriptor in file [BLEMouse.cpp](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/src/BLEMouse.cpp) make sense, is a mouse hid descriptor. I need develop a Gamepad and need help with descriptor and implementation of messages.
I´m studying a descriptor in Arduino SDK of Nordic ([folder link](https://github.com/NordicSemiconductor/ble-sdk-arduino/tree/master/libraries/BLE/examples/ble_HID_template))
I need help with a library to HID gamepad. 
thanks

RezApp, 2020-08-09:
> I deploy to my board the example [HID_joystick_mouse.ino](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/HID/HID_joystick_mouse/HID_joystick_mouse.ino). In Android [...]

RezApp, 2020-08-09:
I have the same question
@marcotchella, did you find an answer for this?