sandeepmistry/arduino-BLEPeripheral #233: HID not working in nrf51

arunmagesh, 2018-05-25:
Hello,
I have flashed my nrf518xxaa module with s130.

Whenever I flash HID keyboard or mouse. it just broadcasts and connects and the keystrokes are not actually passed to the device. i can see all the HID serv and charac in nrf tools .

What could be wrong??

arunmagesh, 2018-05-25:
`// Copyright (c) Sandeep Mistry. All rights reserved.
// Licensed under the MIT license. See LICENSE file in the project root for full license information.

// Import libraries (BLEPeripheral depends on SPI)
#include <SPI.h>
#include <BLEHIDPeripheral.h>
#include <BLEKeyboard.h>

//#define ANDROID_CENTRAL

//custom boards may override default pin definitions with BLEHIDPeripheral(PIN_REQ, PIN_RDY, PIN_RST)
BLEHIDPeripheral bleHIDPeripheral = BLEHIDPeripheral();
BLEKeyboard bleKeyboard;

void setup() {
  Serial.begin(9600);
#if defined (__AVR_ATmega32U4__)
  while(!Serial);
#endif

  // clear bond store data
  bleHIDPeripheral.clearBondStoreData();

#ifdef ANDROID_CENTRAL
  bleHIDPeripheral.setReportIdOffset(1);
#endif

  bleHIDPeripheral.setLocalName("HID Keyboard");
  bleHIDPeripheral.addHID(bleKeyboard);

  bleHIDPeripheral.begin();

  Serial.println(F("BLE HID Keyboard"));
}

void loop() {
  BLECentral central = bleHIDPeripheral.central();

  if (central) {
    // central connected to peripheral
    Serial.print(F("Connected to central: "));
    Serial.println(central.address());

    while (central.connected()) {
      {
      

        bleKeyboard.print("TESTING");
      }
    }

    // central disconnected
    Serial.print(F("Disconnected from central: "));
    Serial.println(central.address());
  }
}`

jpiat, 2018-10-25:
I have exactly the same problem. Did you solve the issue ?

hannesweisbach, 2021-05-13:
I experienced this issue under macOS. The sample code worked for Android. The sample code is however not a complete compliant BLE HID device. There is a lot of stuff missing (Battery Serivce, Device Information, …) which I added manually and finally got it working. I haven't figures out yet what the minimum set of services/characteristics is that macOS accepts as a valid input device.

I've also noticed that there is still some things with the bond store: I need to remove the BLE device from macOS and reconnect to make it work. I believe the storing keys to the bond store does not quite work right for me currently. When macOS re-connects the device does not have the correct keys. After removing the device from macOS and re-bonding everythings works again.

macvenez, 2023-02-14:
> I experienced this issue under macOS. The sample code worked for Android. The sample code is however not a complete compliant BLE HID device. There is a lot of stuff missing (Battery Serivce, Device [...]

Hi, can you share your working sketch with those added characteristics? Thanks!

hannesweisbach, 2023-02-22:
Hey!

No, I haven't kept the code I don't think. It's pretty easy though.

macvenez, 2023-05-13:
I tried changing many settings about those characteristics but I can't find any combination that works correctly.
The NRF51 is seen correctly as a keyboard but can't send any keystroke; furthermore the device can't be edited on Android (settings like keyboard layout, where usually you can with commercial keyboards), seems like it's missing some sort of security or mandatory characteristic to be recognized as a trustable HID device.
If you can help me retrieve this sketch or figure out how it was working for you it will be amazing.
Still thanks