sandeepmistry/arduino-BLEPeripheral #234: putting peripheral to sleep between BLE updates

lyakh, 2018-05-27:
Hi Sandeep,
Thanks for your work on nRF52832 nd BLEPeripheral! I'm using an nRF52832 breakout board from SparkFun, managed to enable the Arduino LowPower library for it, so I can now put it so sleep similar to the originally supported by the library Primo boards. Now I'm trying to put the board to sleep between BLE sessions, i.e. connect to a central, report whatever data the board has to send, go to sleep, wake up, reenable BLE, report new values, etc. I cannot seem to reenable BLE after waking up. Do you have any ideas what the correct way to achieve that?
Thanks
Guennadi

pfodApp, 2018-10-25:
Can I second this request. Currently BLEPeripheral uses poll() in the loop().  
This uses too much current.  Redbear Nano implementation lets you use
 ble.waitForEvent(); // just sleep 
in loop() instead
and just handle the BLE stuff in call backs.
I realize this is a structural change to BLEPeripheral but is necessary for low current usage

KEranda, 2020-10-11:
@lyakh @pfodApp Did you guys ever figure this out. I'm trying to do something similar. Mine just doesn't seem to sleep while BLE is enabled.

drmpf, 2020-10-11:
My solution was to hack Sandeepmistry's nRF5 support for  nRF52832 chips to get very low power
see [Easy Very Low Power BLE in Arduino -- Part 1](https://www.forward.com.au/pfod/BLE/LowPower/index.html) and parts 2 and 3
However Sandeepmistry's lastest support for Nano33BLE is based on mbed rtos and so is completely different.  Not sure what would be needed to make that very low power.  
Sandeepmistry can you look at adding very low power support for those boards?.