sandeepmistry/arduino-BLEPeripheral #236: Is it possible to do wired UART?

marcsza, 2018-06-14:
I started developing some applications just to feed my curiosity about BLE and nrf52. With this library, is it possible to implement wired serial communication as well as scannng for beacons?

My goal is to scan for a beacon and as soon as i find it i send serial (wired) info to an arduino board. Is it possible?

Regards,
marcsza

francesco98, 2018-06-23:
Using SoftwareSerial, I receive this error: `fatal error: util/delay_basic.h: No such file or directory`

don, 2018-06-29:
@francesco98 maybe your board isn't supported or you need something extra. See if other people using your board have a solution. Or use one of the UARTs built into the board.

HalFrgrd, 2018-07-01:
I'm having some trouble getting uart communication between an nrf51822 and a pro micro. What is the purpose of the Uart.h/Uart.cpp files in the arduino-nrf5 core? How would I use the setpins method in those files to get uart on pins 24,25?
I have this running on my nrf51822:
```C
void setup() {
  Serial.begin(9600);
}

void loop() {
  delay(1000);
  Serial.write("testing");
}
```

and this running on a pro micro that is connected to the nrf51822:

```C
void setup() {
  Serial.begin(9600);
  Serial1.begin(9600);
}

void loop() {
  if (Serial.available()) {      // If anything comes in Serial (USB),
    Serial1.write(Serial.read());   // read it and send it out Serial1 (pins 0 & 1)
  }

  if (Serial1.available()) {     // If anything comes in Serial1 (pins 0 & 1)
    Serial.write(Serial1.read());   // read it and send it out Serial (USB)
  }
}
```
Should this be enough to see 'testing' on the Serial Monitor?

Thanks, 
Hal

ankitmax, 2021-03-04:
> I'm having some trouble getting uart communication between an nrf51822 and a pro micro. What is the purpose of the Uart.h/Uart.cpp files in the arduino-nrf5 core? How would I use the setpins method [...]

Hello,

Did you find solution to this? That is, how to set RX and TX pins for nrf51822 UART? And use Arduino board to communicate with Nordic UART wires interface?

HalFrgrd, 2021-03-04:
I can't remember sorry.
I think I may have ended up using the Nordic SDK instead.