sandeepmistry/arduino-BLEPeripheral #241: Android don't work with this libraries

Egauss, 2018-08-08:
Hello, I am using sparkling nrf52832, ble400, boards with this libraries, I can make it work with serial monitor app but when I am trying pair with phone it's saying pairing rejected by "device". Tryed with 10 different androids all examples, iPhone can't see at oll my device in Bt connection. Any advice? By the way android have bluetoorh4.1

paveljanecek, 2018-08-09:
Can you post the code?

Egauss, 2018-08-09:
Examples/temp sensor -no modifications try pair with android

austinfennacy, 2018-08-10:
I'll second this, iPhones can connect and the Androids I tested can't. I just tested `serial.ino` on an nRF52DK that has had SoftDevice S132 successfully flashed and have attempted pairing with both an Android 8.0 and 5.0, and when connecting to `nRF Toolbox` -> `UART` -> `Connect`, I get a message saying `The device does not have required services.` 

I checked the same procedure using a friend's iPhone 7 and after swiping right to expose a serial terminal style view in nRF Toolbox, I sent and received messages from the Arduino serial terminal and the iPhone 7:
[iPhone 7 nRF Toolbox screenshot](https://user-images.githubusercontent.com/33964526/43979615-2e60bc92-9ca0-11e8-8e54-b5d4c863b641.png)

Oddly enough, when using Android 8.0 I'm able to bond using nRF Connect, and it shows these services (?), but I'm not able to send or receive data from the phone:
[Android 8.0 nRF Connect screenshot](https://user-images.githubusercontent.com/33964526/43979952-40e829d0-9ca1-11e8-95cc-4a9286a599fb.jpg)

When viewing the same screen in nRF Connect on the iPhone, I'm greeted first with the services screen:
[iPhone 7 nRF Connect Services screenshot](https://user-images.githubusercontent.com/33964526/43980330-a4988b22-9ca2-11e8-94ba-51839f2f3a6b.png)
Clicking through shows only the 2 relevant characteristics:
[iPhone 7 nRF Connect Characteristics screenshot](https://user-images.githubusercontent.com/33964526/43980331-a4b02e26-9ca2-11e8-8bd5-2b2ce5f4de58.png)

I've also checked with Play Store's "Serial Bluetooth Terminal" app with no luck connecting.

Not sure what is causing this, moving forward I think I'll try tweaking code in `BLEPeripheral.h` since this seems to be a problem with all the examples I have tried.

Any ideas how to get Android to view `Nordic UART RX` and `Nordic UART TX` characteristics?

Egauss, 2018-08-11:
I think maybe I found answer, the soft devices is not compatible, you can find this on softdevice documentation  if library was made for v2 and now we using v6 we are here like is, as simple man who write code can't answer why library dosnt work

Egauss, 2018-08-16:
@austinfennacy I want to say softdevice documentation saying what different versions not supported,  to use that you need update softdevice and recompile, I think this library was made for one version of softdevice and didnt work as need update library code, with Keil just recompile.

bossesand, 2018-08-16:
> @Egauss <https://github.com/Egauss> it's not apparent to me that the
> softdevices were ever patched to support Android in the documentation for nRF52832
> SoftDevice versions
> <https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52832>,
> which makes me think that they are intended to support Android. What gives
> you the idea that S132v2 doesn't support Android and S132v6 does?
>
> It looks like a user had problems with Android in issue #136
> <https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/136> back
> in 2017, and that issue is still open so I'm not sure if this
> core+softdevice+library have ever worked for Android... @bossesand
> <https://github.com/bossesand> were you ever able to get your code
> working with Android?
>
> @sandeepmistry <https://github.com/sandeepmistry> I'll do my best to take
> a stab at this problem but before I go down the wrong rabbit hole, do you
> have any guesses why this core+softdevice+library are unable to connect to
> Android, or which part of the chain is failing? In a comment above I
> documented how iOS vs Android view services/characteristics after the
> serial.ino example is uploaded, if that's of any help.
>
> I'd love to help get this figured out

I tried so many ways so I do not remember what worked or not.
But my first successes was using android and serial to test concept.

MY requirement is to run using Linux, windows and OSX Also.

I am currently using the adafruit libraries for blueprint to connect the
serial using node.js at host.
Have not tested the js on android, only the ble terminals.

Bosse

Egauss, 2018-08-16:
Serial monitor on Android works as no need pairing

bossesand, 2018-08-16:
Not blueprint Bluefruit it should be.
I hate spell checker on android.

Den tor 16 aug. 2018 08:28Bo-Erik Sandholm <bossesand@gmail.com> skrev:

Mboudali, 2023-07-28:
Any update on this issue?