sandeepmistry/arduino-BLEPeripheral #243: Transmit Data As Fast As Possible

webzoid, 2018-08-19:
I've set up a UART characteristic and I'm wanting to stream data from a sensor via the BLECharacteristic.

The UART (and TX/RX) characteristics are defined as follows:

```
BLEService uartSrv("6E400001B5A3F393E0A9E50E24DCCA9E");
BLECharacteristic rxChar("6E400002B5A3F393E0A9E50E24DCCA9E", BLEWriteWithoutResponse, 32);
BLECharacteristic txChar("6E400003B5A3F393E0A9E50E24DCCA9E", BLENotify, 32);
```
And I send data out via:

`txChar.setValue((const unsigned char *)&data, sizeof(data));`

What I've noticed is that after about 5/10 successful transmissions, the BLE engine seems to stop and no more data is transmitted.

Am I doing this correctly? What mechanism can I implement to ensure that the data is successfully sent before I can start another transmission?

dots123, 2018-08-25:
Hi webzoid,

Have you managed to solve the problem? 
I was looking in BLESerial.cpp and they flush the data after they send some chars. Take a look here: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/serial/BLESerial.cpp

webzoid, 2018-08-25:
Hi @dots123 

I haven't yet solved the issue. I call `poll` (as in the BLESerial.cpp `flush` function) after the `txChar.setValue(...)` function yet this doesn't seem to make a difference.