sandeepmistry/arduino-BLEPeripheral #244: How to change MAC ID in NRF52

Autobot86, 2018-08-23:
hi friends am the beginner of nrf52832. i try to code for simple iBeacon i have doubt as what is the use of line " unsigned short measuredPower = -55;" ?? i find there is no change after assign values to it and how to edit device macid in that code.

Autobot86, 2018-08-24:
SVCALL(SD_BLE_GAP_ADDRESS_SET, uint32_t, sd_ble_gap_address_set(uint8_t addr_cycle_mode, ble_gap_addr_t const *p_addr));

i see this function call in soft device library how to call this function in iBeacon example code to change my device address

Autobot86, 2018-08-24:
i try to change device address using this function

  static ble_gap_addr_t m_central_addr;
  m_central_addr.addr_type = BLE_GAP_ADDR_TYPE_RANDOM_STATIC;
  m_central_addr.addr[0] = 0xfd;
  m_central_addr.addr[1] = 0x71;
  m_central_addr.addr[2] = 0x5f;
  m_central_addr.addr[3] = 0x45;
  m_central_addr.addr[4] = 0x2b;
  m_central_addr.addr[5] = 0xc6; // 2MSB must be set 11
  
  for (int i = 0; i < 5; i++)
    Serial.println(m_central_addr.addr[i]);

  sd_ble_gap_address_set(&m_central_addr);

but error come how can i change device address in my NRF52 board in arduino code

AnthonyAchoy, 2019-12-27:
Could you solve it?

AnthonyAchoy, 2019-12-27:
I can actually compile but it doesn't modify the device address, I tried calling the funcion in many places, but it just print the elements of the array.