sandeepmistry/arduino-BLEPeripheral #250: Bondstore

suvodip1212, 2018-09-27:
Could some private data be kept in flash using bondstore? I tried using the read command but failed to do so(for now). Any suggestions?

h2zero, 2018-10-12:
Also interested in doing this. I’ve already modified it to store IRK/LTK’s. So I think this would be simple enough to implement.

suvodip1212, 2018-10-17:
Pls share an example. Maybe it will help?

h2zero, 2018-10-17:
Sure, when I get a few mins I'll make a fork and upload the code.

h2zero, 2018-10-22:
So I made a fork but then I couldn’t find the code with the bond store changes... I must have overwritten it when I was moving files around :(.

I guess I’ll redo it and I might as well  write some code to store arbitrary data as well.

Sorry for the delay.

dots123, 2020-12-04:
> So I made a fork but then I couldn’t find the code with the bond store changes... I must have overwritten it when I was moving files around :(.
> 
> I guess I’ll redo it and I might as well write some code to store arbitrary data as well.
> 
> Sorry for the delay.

Hi! Can you share an example for storing that bond data in flash? Or can you give an example of how to write data in flash while the BLE is running?

h2zero, 2020-12-18:
Wow, so sorry, I completely forgot about this. What I ended up doing is just creating a second instance of `bondstore` and storing arbitrary data in it by just using `bondstore::putData()`. I had modified the code previously for other things and made another solution but ended up doing it this way instead.