sandeepmistry/arduino-BLEPeripheral #258: Upgrading to Latest SoftDevice and SDK

webzoid, 2019-01-11:
Does anyone know of a fork of this repository which may have been updated to support the latest Nordic softdevice and nRF52 SDK? The data throughput of this earlier softdevice is pretty horrid (I'm seeing about 4kbps MAX) whereas later versions of the softdevice/SDK have shown up to 1300kbps is possible:

https://www.novelbits.io/bluetooth-5-speed-maximum-throughput/

I'm still trying to get my head around the nRF52 SDK (and latest softdevice API) before I attempt such a feat but just wondered if such a thing already exists.

zanimax12, 2019-02-18:
@webzoid Did you found a way to use the latest softcore?

webzoid, 2019-02-18:
@zanimax12 Unfortunately I have not. In the end, I tried moving over to the Adafruit Bluefruit libraries which use the latest softdevice, however, performance was nowhere near what I'd hoped for - so far as to say that the performance of the BLE UART service was worse that this BLEPeripheral library using an older softdevice. 

Currently, I'm stuck with an old softdevice and suffering the consequences of poor performance. Maybe I should just bite the bullet and get my head around the NRF SDK using an Eclipse platform rather than the Arduino platform.