sandeepmistry/arduino-BLEPeripheral #259: Add support for nRF52840

tamberg, 2019-01-18:
Hi Sandeep,

is there a chance your library already works with the nRF52840?

Would love to use it with https://www.adafruit.com/product/4062

Kind regards,
Thomas

tamberg, 2019-01-18:
Adafruit writes:

> Since the underlying API and peripherals are the same for the '832 and '840, you can supercharge your older nRF52832 projects with the same exact code, with a single recompile!

melvyniandrag, 2019-01-23:
I'm not sure this qualifies as an issue, this is a question. I don't have this chip to experiment with, so I can't help you out. Were you able to use the library?  Looks like it should work, right?

I'm new to this BLE stuff, maybe you are too?

dsteinman, 2019-05-27:
Unfortunately it looks like this library does not work on the nRF52840.  I receive this error while compiling:

```
Arduino: 1.8.9 (Mac OS X), Board: "SparkFun Pro nRF52840 Mini, s140 6.1.1 r0, Level 0 (Release)"

In file included from /Documents/Arduino/libraries/BLEPeripheral/src/utility/lib_aci.h:34:0,
                 from /Documents/Arduino/libraries/BLEPeripheral/src/nRF8001.h:9,
                 from /Documents/Arduino/libraries/BLEPeripheral/src/BLEPeripheral.h:26,
                 from /Documents/Arduino/NRF40-BLEtest/ble.ino:2:
/Documents/Arduino/libraries/BLEPeripheral/src/utility/hal_platform.h:75:6: error: #error "Unsupported platform"
     #error "Unsupported platform"

      ^
exit status 1
Error compiling for board SparkFun Pro nRF52840 Mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

I think it would be a worth a shot to edit [hal_platform.h](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/src/utility/hal_platform.h).  Just one line change might be needed:

```
#if defined (__AVR__) || defined(__SAM3X8E__) || defined(__SAMD21G18A__) || defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MKL26Z64__)
```

Unfortunately I don't know what to add there.